gtkwave/gtkwave #129: Continuous plotting

suzizecat, 2022-06-09:
Hi there !

I don't know how much work that would be, but would it be possible to trace (FST or other) waves "continuously" as they are dumped by the simulator ?
The idea comes from tools such as simvision which are able to plot "in real time" the data as it is generated by the simulator.

An example of usecase cold be the "interactive" simulation capabilities that I discovered with cocotb.
The idea is simple, cocotb uses python testbenches, and therefore, you can use the `input` function to prompt the user.
It could be interesting to see the wave generated just after (each) user input. as they go.

This idea came to me also because when I load a FST waveform that is being written, GTKWave "see" that the file is not finished and retry to load until the simulation is done. Would it be possible to plot the available data each time GTKWave tries to load ? 

Anyway, thanks for this great tool !
Best regards,
Julien FAUCHER

NourFr10, 2022-06-10:
Hi, based on the gtkwave documentation, I think that you can do interactive simulation with VCD files but not others like FST.

> GTKWave is an analysis tool used to perform debugging on Verilog or VHDL 
simulation models. With the exception of interactive VCD viewing, it is not 
intended to be run interactively with simulation, but instead relies on a postmortem approach through the use of dumpfiles.

gtkwave, 2022-06-10:
VCD is the only loader that supports continuous plotting.  That required a special loader for VCD in addition to the regular one.  It works across all signals in a file, but its original intent was that you might dump an interface or some subset of signals and can look at those over very large time ranges.

Note that nothing prevents reading from block-based files (FST, FSDB, LXT2, VZT) while the trace is being generated by sim, but you will have to manually reload it.
-Tony