gtkwave/gtkwave #130: Moving towards GTK4

rfuest, 2022-06-10:
After a long pause I finally resumed working on my rewrite of the GtkWave drawing code. I'm getting closer to the point where I want to create a PR for my work, but the closer I get the more I'm wondering what the best way to integrate my changes into GtkWave is.

I haven't tested my code with GTK2 for a long time and I could either start to work on getting GTK2 to work, but IMO it would be more useful to work towards GTK4 support and don't spend any more time on GTK2. GTK4 is supposed to work much better on macOS, which should remove the need to keep GTK2 support working.

My suggestion would be to start a new branch for GtkWave 4 and merge my changes to that branch. Merging my changes into the current version of GtkWave immediately might not be a good idea regardless of the GTK2 support, because the changes are extensive and not tested very well. And the branch would be a good place to modernize the GTK3 code to make the transition to GTK4 easier, like using only gestures for input and migrating to actions and menu models.

gtkwave, 2022-06-10:
I'd say you can safely abandon any of the gtk2 support.  If you want to do a new branch, or subdirectory like with the other versions, go for it.
-Tony

NourFr10, 2022-06-11:
is it possible to add plug-in system to the gtkwave3-gtk4 ? so we don't need to change the source code for adding new features ?

rfuest, 2022-06-11:
> is it possible to add plug-in system to the gtkwave3-gtk4 ? so we don't need to change the source code for adding new features ?

Adding a plug-in system is not an easy task and isn't on my short term goals list for the GTK 4 port. But the code changes should at least make it easier to add features to GtkWave by changing the source code.

gtkwave, 2022-06-11:
Manchester tried a plug-in system for gtkwave-2.  Yes, there was such a thing as a gtkwave 2--about 22 years ago I was threatened with being fired for working on gtkwave by my employer so I turned over the codebase to someone else.  That worked pretty good until the main developers got their Ph.Ds and moved on.  I resumed 3.x from 1.3 after they stopped development.  There are/were actually some commercial tools that forked off the 2.x branch and screenshots show their heritage if you know what you're looking for--specifically their version of the toolbar at the top of the window.
https://apt.cs.manchester.ac.uk/ftp/pub/amulet/balsa/other-software/
As far as plug-in systems go.  It's a good idea, and it worked well for what Manchester wanted, but supporting that one across OSs is non-trivial.  I don't know if glib supports a method to do it across OSs, but if not, it's added complexity that doesn't need to be there (otherwise I would've ported it forward to 3.x!) so I agree with Ralf on this one.

-Tony

NourFr10, 2022-06-12:
I don't see any plug-in system in the Manshester's forked version of gtkwave in the link you attached.

gtkwave, 2022-06-13:
All the loaders are plug-in based using a well-defined DLL format for loaders.  It's not some abstract mechanism for function, but is specific to loaders.  Development stopped before it moved beyond what was implemented.  I don't remember if they got around to having some sort of spec file or whatever to define what loaders were available--probably not.

-Tony

NourFr10, 2022-06-18:
How about tcl DLL with the current version of gtkwave ? are those not plugins ?

gtkwave, 2022-06-20:
There was a developer about 15 years back who did make Tcl as a dynamic loader and there are ./configure options to support it.  I forgot about that as I've never had to used it.

-Tony

closed 2022-08-06 by rfuest

NourFr10, 2022-08-11:
Hi, @rfuest 
could you please tell me the instructions to be able to build this version (gtkwave4) ? I tried ./autogen ./configure ... but that didn't work. I tried even the meson build system, but I had  this error `meson.build:1:0: error: value "gnu+11" (of type "string")`

rfuest, 2022-08-11:
You need to use meson the autotools build doesn't work in the gtkwave4 directory. There was a typo in the `meson.build` file which caused the error you described. I've just committed a fix.

NourFr10, 2022-08-11:
still have this issue :
`meson.build:30:0: ERROR: Dependency "bzip2" not found, tried pkgconfig and cmake`

rfuest, 2022-08-12:
You seem to miss the development package for `bzip2`. Which OS are you using? Under Ubuntu you would, for example, need to install `libbz2-dev`.