gtkwave/gtkwave #139: xml2stems creates incomplete .stems file

supleed2, 2022-07-07:
This was tested with a fresh copy compiled using the `gtkwave3-gtk3` folder in commit https://github.com/gtkwave/gtkwave/commit/49a2a53caee83890dff503c15815fb53d5ccde74.

When generating the stems file, the filepath and line numbers are missing on module lines, eg:
`++ module test_DUT file lines 0 - 0` (2 spaces between file and lines, double spaces don't show in markdown)

I was also wondering what the format for the stems file was and if there is a formal/informal specification, I didn't see one mentioned in #53, just the `fprintf` lines., as I was trying to put together a (temporary?) python script to do the conversion.

Once I have it working I could have a go at converting it to C/C++, but my confusion is mostly around the "uniquified" names Verilator gives to parameterized modules.

The component line: `++ comp instanceName type moduleDefName parent parentDefName` uses the "type" or defined name to indicate the parent module, which may not be unique, instead of the instance hierarchy name. This also breaks when using the Verilator "uniquified" names as then RTLbrowse fails to find the module definition, even when given the correct line number.

Let me know if you need any extra explanation or a small example, I can try and put one together.

gtkwave, 2022-07-11:
Unfortunately, right now I don't have time to look at any of this as I'm too busy with the day job.  It's been quite a while since I've had to use rtlbrowse.  At this point it needs a lot of work to support constructs such as generate loop indices.

The long and short of how this file works is it's supposed to provide the module definition locations as well as instantiatiations so a recursive descent can regenerate the original hierarchy to correlate the signals found in a file like FST with the sourcecode.

Run the des.v toy example from examples/ through it and it should be fairly obvious how it works.  I think the ending line number is optional, mainly because there will be tokenizing in rtlbrowse until endmodule or something similar.
-Tony

jotego, 2023-10-03:
I confirm that stem files are also useless in commit c05122b (tag v3.3.116). You are soon shown a **MISSING** label next to each module, regardless of all the files existing in the disk and correctly mentioned in the XML file.

[image]