gtkwave/gtkwave #140: why still using C and GTK for GTKWave ?

NourFr10, 2022-07-22:
Hi Tony,
I started development on gtkwave four month ago but I found myself coding bunch of lines just to add a simple features to gtkwave.
I am just wondering why not start using C++ and Qt for more flexibility during development  over the gtkwave project.
this will encourage more developers to contribute to GTKWave.
Thanks.

rfuest, 2022-07-22:
I'm not Tony, but I would like to give my opinion on this subject. I've started to contribute to GTKWave a while ago and can understand some of your criticism. The age of the codebase is noticeable and the tight coupling between different parts of GTKWave can make it difficult to change things.

But changing to C++ and Qt isn't the solution for this problems IMO. Changing from GTK to Qt means that the entire UI would need to be rewritten. GTK 4 is a big improvement over GTK 3 and a much better target for further development of GTKWave.

My plan for the GTK 4 port of GTKWave is to slowly extract the backend parts of the current codebase into a library. By staying with C and GTK this can be done step by step in small parts. I've actually started with this refactoring and it will improve the developer experience without throwing away the great work done by Tony and others, like changing the language would.

gtkwave, 2022-07-23:
It's still using C and GTK for various reasons...
Rewriting that much code from scratch is a lot of effort.  If I were a manager and had to size it for schedule and budget, I would estimate it at 5-6 PY (person years).  An architectural specification for it that a team could work from would be in the hundreds of pages, mainly because there are thousands of pieces of function that slowly collected over time.  Someone could probably code up a basic strawman viewer in two to three weeks, then the real "fun" would begin what with adding all the things that would take quite a long time to get to feature parity, and the effort would be abandoned.  I've seen this at least three times in the past on mailing lists and newsgroups (remember those?) where someone would say, "Arrrrrgh, gtkwave drives me nuts!  I'm writing my own viewer!"  Six months later, abandonware.  Lather, rinse, repeat.

Part of the reason for a lot of effort required is a wave viewer is more than a GUI.  Many unglamorous things happen behind the scenes that are invisible, such as managing enormously large data structures efficiently so the viewer doesn't bog down at typical zoom levels.  A lot of this code was developed by trial and error over the years so it works, but wow is there cruft.  Some of the real crazy stuff was to support 32-bit memory limitations.  "I loaded a 1GB VCD file and the viewer crashed.  Why?"  Oddly enough, things like that set me on the path to developing FST where I more-or-less rediscovered the concepts behind FSDB.  I have yet to see anyone else develop any open source sim dump formats.  Today if I was just starting from scratch, I'd build a new viewer around a database format like FST.  Lesson learned, but such things didn't exist in open source in 1998, so I couldn't have done it anyway.
 
New code for all the non-GUI function almost certainly would not be as stable.  There's something to be said about "old and boring" when I use the viewer daily for my day job.

GTK4 is much more reasonable to port to than Qt as it doesn't require a from scratch rewrite to deal with a radically different toolkit architecture.  I was actually going to do the GTK4 rewrite on the heels of GTK3 but the (3.9?) toolkit was too much of a moving target back then, otherwise it would have already been done.

5-6 PY doesn't sound like a lot when the codebase is pushing 25 years old, but I'm far too busy with my day job and I just don't have the time these days for spare time work except for one-off feature adds that help with work.  If you look at the change log from 2012 on, most of the adds were either user contributions or were practical ones to aid my RTL development on various Zen 1, 2, ... instruction fetchers, for example, supporting native loading of FSDB and VF files as that's what comes out of VCS or emulation at work.
I definitely agree with Ralf's direction of refactoring.  

-Tony

NourFr10, 2022-07-26:
Well, first I am a fun of this wonderful and open source project, and I am not criticizing it, but I asked that question to discuss the feasibility of this idea with you. 

Yah I agree with you both, and it seems not an easy task to rewrite all those pieces of code with another OOP language like C++, but as Tony said It would be great idea  if we start a new viewer around the database format like FST, and then we start slowly supporting other formats (VCD and others). much the UI can easily be rewritten with other cross-platform software like Qt, for the back-end parts it could be slowly rewritten over time. after 3-4  years we will have another GTKWave that make the same value as the current but it will make things easy to change and will have more flexibility and will improve developer experience.

Tony and others made a great work and I can see that base on my four months internship experience with GTKWave and I appreciate their work.

rfuest, 2022-07-26:
I still think that C++ isn't necessary, but I might be biased because I don't have a lot of experience with C++ and never really liked the language. But I also want to use some OOP in my refactor. GObject, which is the base for GTK, makes it possible to implement classes and interfaces in plain C. There is a bit more boilerplate involved, but once you wrote a few classes it isn't very complicated to use. It also fits in great in the GTK ecosystem and you can use the GTK inspector to inspect and modify the values of any property of a GObject. In my experimental drawing rewrite branch you could, for example, change the zoom factor of the wave window in the GTK inspector and the entire UI would update automatically. And with GObject Introspection it is possible to generate bindings for a lot of other languages: https://gi.readthedocs.io/en/latest/users.html These features should make it easier implement plugin support later.

@NourFr10 I'll try to create a first PR with some of the initial changes I want to make soon. This should give you and also Tony a better idea of what I'm trying to achieve. If you like the direction it is going and you are not to fixated on QT and C++ perhaps you could contribute to that branch. Any help would be appreciated.

At the moment I'm thinking about changing the `TraceEnt` struct into an abstract class. Each type of trace (hptr/vptr, digital/analog...) would be a separate class which inherits from the abstract base class. I think this could be used to bundle a lot of code related to the different trace types into a single place instead of handling the different trace types all throughout the code.

A more advanced change would be to move from a traces list to a traces tree. If each trace could have children it would be easier to implement groups or dynamically create child traces for `TTRANSLATE` traces (without using blanks). But it would also need changes to the save file format and the TCL API.

NourFr10, 2022-07-28:
I have given C++ as an example, as OOP is supported natively and more suitable for managing the complexity, I wanted to suggest a decoupled and functional pieces of code and easily changeable, but I am not fixated to it if the GObject could brings these features to C and GTK.

Yah I am interested with the plugin system, I was searching for it four months ago, but as GTKWave does not have this feature, I started changing source code directly.

The idea of changing the TraceEnt to an abstract class is good, that will be as simvion i think, but honestly that will require an advanced changes within GTKWave as the TraceEnt is used approximately everywhere and other data structure depends strongly on it, I can help with that.
 
Another idea I want to suggest is to generate TCL file instead (or optionally) of save file (simvision does this),  I was thinking for this a while when I had to regenerate some debug traces, but a I didn't even tried. so continued working on save file.

rfuest, 2022-07-28:
> The idea of changing the TraceEnt to an abstract class is good, that will be as simvion i think, but honestly that will require an advanced changes within GTKWave as the TraceEnt is used approximately everywhere and other data structure depends strongly on it, I can help with that.

Yes, this will require changes to the entire GTKWave codebase and is a lot of work. But I've already started with this refactor as a two step process. The first step, I'm currently working on, is to change the code to not access the `TraceEnt` fields directly. For example by changing code like `t->flags |= TR_HIGHLIGHT;` to `gw_trace_set_slected(t, TRUE);`. This can be done gradually, because code that directly accesses the fields remains working. Once the other code is changed `TraceEnt` can be converted to a class without requiring further massive changes.

> Another idea I want to suggest is to generate TCL file instead (or optionally) of save file (simvision does this), I was thinking for this a while when I had to regenerate some debug traces, but a I didn't even tried. so I continued working on save file.

I had the same idea a while back, but came to the conclusion that using TCL is not a good idea. IMO a save file format shouldn't allow arbitrary code to run for security reasons. Maybe this could be addressed by running the TCL code in a sandbox and using a timeout, but I think it's better to keep scripting and save files separate.

I think it would be a better idea to use a serialization format like JSON or S-Expressions to replace the current save format with a more readable and extendable one. I personally prefer the S-Expression format, that is for example used by KiCad, over JSON ( https://dev-docs.kicad.org/en/components/sexpr/ and https://dev-docs.kicad.org/en/file-formats/sexpr-intro/), because it's easier to read and to edit by hand. But JSON has the advantage that it's much more common and there are a lot of libraries to read and write JSON files.

gtkwave, 2022-07-28:
TCL as a save file sounds like a good idea in theory, however trace imports from a dumpfile would have to be reworked some.

For performance, you want to bunch up everything that goes into an FST, FSDB, or whatever file, then make a query for multiple signals all at once so you have a single pass through the dumpfile database.  If the TCL file is of the kind that runs from top to bottom and is done, it would work well, but for something that runs continuously as a first class TCL script, it would be have perf issues.  Again, not impossible, but it would need some optimizations.  For example

gtkwave::dumpFileSignalAddStart... signal adds go here and could be generated via arbitrary looping algorithms, etc. ...
gtkwave::dumpFileSignalAddEndgtkwave::dumpFileSignalImport
Real easy, but not initially obvious because it's a perf detail with respect to implementation as users never really have to be concerned with such internals.

-Tony

NourFr10, 2022-08-01:
> Yes, this will require changes to the entire GTKWave codebase and is a lot of work. But I've already started with this refactor as a two step process. The first step, I'm currently working on, is to [...]
So when the first phase is done, we have just to change the body of `gw_trace_set_slected(t, TRUE)` as well as the  `TraceEnt ` to an abstract class. that would be nice and smart. I can start collaborating with you by end of September if that would not be too late because I have to finalize my graduation internship here in France. 

> My plan for the GTK 4 port of GTKWave is to slowly extract the backend parts of the current codebase into a library.
that would be like kind of API for gtkwave4, that doesn't depends on any GLOBALS variables is that correct ?

NourFr10, 2022-08-01:
by the way I want to know if it is possible to add child to trace and make it hidden by default and when we click left twice we could see all the children that has been add. I know that `bvptr combine_traces(int direction, Trptr single_trace_only)` combines traces and return a vector but for my case I don't want to changes data (on-screen state) of the parent trace.
this what would be shown when clicking on expand but the transaction data must not be changed. 
[image]

rfuest, 2022-08-01:
> So when the first phase is done, we have just to change the body of `gw_trace_set_slected(t, TRUE)` as well as the `TraceEnt ` to an abstract class. that would be nice and smart. I can start [...]

There will be plenty of work left by end of September.

> that would be like kind of API for gtkwave4, that doesn't depends on any GLOBALS variables is that correct ?

Yes, that is correct. Removing `GLOBALS` is one of my main goals for gtkwave4. Code inside `libgtkwave` doesn't have direct access to `GLOBALS` anymore.

rfuest, 2022-08-02:
> by the way I want to know if it is possible to add child to trace and make it hidden by default and when we click left twice we could see all the children that has been added. I know that `bvptr combine_traces(int direction, Trptr single_trace_only)` combines traces and return a vector but for my case I don't want to changes data (on-screen state) of the parent trace.
> the picture attached illustrate what would be shown when clicking on expand but the transaction data must not be changed.
> this children will be added dynamically after applying a filter for the transaction that I have already developed, and now I need just this features.

I assume the filter you are talking about is custom code, which creates traces and not using the "Transaction Filter Process" feature?

If that is the case you should be able to implement like this:
- The parent trace needs to additionally have the `TR_GRP_BEGIN` and `TR_CLOSED` set.
  And set `t_match` to point to the group end trace.
- For the child traces you need to set `t_grp` to point to the parent trace
- You need to add a group end trace after the last child trace:
   `flags = TR_BLANK | TR_GRP_END | TR_COLLAPSED` & set `t_match` to point to the parent trace

NourFr10, 2022-08-04:
> I assume the filter you are talking about is custom code, which creates traces and not using the "Transaction Filter Process" feature?
>
> If that is the case you should be able to implement like [...]
Yah my filter is costume code (kind of callback function), it traverses the transaction trace (String) and creates two child groups (Req/Rsp), 
Thanks. the three point are clearly demonstrated in `unsigned create_group (char* name, Trptr t_composite)` body so i had just to remove the parent trace before calling it. and call the `void CloseTrace(Trptr t)` over the `t_composite`.

closed 2022-08-10 by NourFr10