gtkwave/gtkwave #145: WAVE_HIERFIX define

rfuest, 2022-08-14:
Is the `WAVE_HIERFIX` define still used or is it part of some old unused code? The code contains `#ifdef`s, but there doesn't seem to be any way to define `WAVE_HIERFIX`, except from modifying the `CFLAGS` manually.

gtkwave, 2022-10-11:
I believe it's obsolete and can be safely removed.  It probably hasn't been compiled in at least since 2009 and definitely not used since then (when FST came online and I make better usage of hierarchy trees).

From the looks of it, it's from back in the day when the original algorithm would sort based on a qsort or merge sort of flattened symbol names rather than recursively sorting levels of the tree.  It's sort of a trick where an alternate identifier was used in place of '.' to force a "reset" of the sort at that boundary by temporarily making the dot character 0x1 then after the sort was complete, it would be reverted back to a dot.

enum VCDName_ByteSubstitutions { VCDNAM_NULL=0,
#ifdef WAVE_HIERFIX
VCDNAM_HIERSORT,
#endif
VCDNAM_ESCAPE };

Regarding VCDNAM_ESCAPE, if(GLOBALS->fast_tree_sort) is typically the norm, not the else blocks that use it.
-Tony

rfuest, 2022-10-11:
Thanks for the explanation. Do you think that the `fast_tree_sort == 0` code can also be safely removed?

gtkwave, 2022-10-11:
Yes.  It hasn't really been activated in years.  I kept it as a fallback plan in case the fast tree sort was broken, and...it's not broken.
-Tony

closed 2023-08-28 by rfuest