gtkwave/gtkwave #148: evcd2vcd handles line feeds improperly

yihuajack, 2022-08-30:
evcd2vcd has problems in handling line feeds, for example:
```
$scope
    module testbench.adder_instance
$end
```
will output
```
$scope
```
```
$var
    port 1 <0 data0
$end
```
will cause
> [1]    270 segmentation fault  evcd2vcd example4.evcd

gtkwave, 2022-08-30:
It's not a real parser and wasn't intended to be by design.  The VCD loaders in the gtkwave executable as well as helper files such as vcd2lxt/vcd2lxt2/vcd2vzt, I believe could handle line breaks in VCD such as in that place, but eventually most of the vcd2whatever loaders/converters at/after the FST file format (that is, vcd2fst) was created moved to a faster read where in cases for scopes and variables

$directive ... $end is contained on a single line.
The evcd2vcd loader is in that camp.  For the most part, it's a bit of a lazy simplification not handling those line feeds like you've shown.

The main reason was that tools such as VCS standardized and simplified the free form format of VCD/VCD-like files a lot compared to what the Verilog spec dictated so there was no need to support arbitrary line breaks.  Commercial implementation supercedes spec conformance in a lot of cases in the industry.  I know that sounds weird, but that's the way things turned out over the years.  In the past, I had AEs from two of the big name companies tell me it was a bit of a waste of time implementing a reader for something the tools never generated, so I didn't.

For most people who have raised VCD loader crash issues over the years, everything points back to that person writing his own VCD writer which emits differently from VCS or NC Verilog.  Any time I get reports of a VCD loader crashing, it's almost always a case where someone made his own VCD writer.  I don't know if that's the case with you or not.  It's annoying for the user, but I don't consider there really to be security implications, mainly because it's not like people download malicious VCD files from the internet.  :-)

I don't remember the last time I tried emitting EVCD from VCS or NC was (then tried to read it).  For all I know, the reader is horribly broken versus what those tools currently emit.  Outside of vectors for chip tester usage, I've never had to touch EVCD, and I think that last time may have been way back in 2005 when I was in Japan and was doing hands on work on a tester for my employer.  The last time it looks like I touched the source for that file evcd2vcd.c converter was 2014.  The first date was 2009, which was after FST was created so it would have used the simplified/faster loader routine.  Sorry about the inconvenience.

-Tony

yihuajack, 2022-08-31:
Thanks for such a thorough explanation. The problems emerged when I was reading 1800-2017 IEEE Standard for SystemVerilog--Unified Hardware Design, Specification, and Verification Language and manipulating some examples for developing an interface of `pyvcd`. In page 660-661 in Section 21.7.2.3 Description of keyword commands, the standard gives two examples:
```
$scope
    module top
$end
```
and
```
$var
    integer 32 (2 index
$end
```
I found that it's true that the main function of `gtkwave` can identify and parse EVCD files properly together with VCD.
I also used ModelSim to generate some EVCD examples. These examples have line feeds in timescale, version, and date declarations but not in var or scope declarations, so the parsing problems will not emerge.