gtkwave/gtkwave #149: Demanded features for GTKWave 4

NourFr10, 2022-08-31:
Hi, 

While performing FST waveform debugging via a standalone C++ app. I have two FST files the first is the one generated by cvc simulator. The second is the one where I dump my custom debugging traces. I use TwinWave to open both files and make comparison. I would prefer that GTKWave had the possibility to open two file within only one session and one viewer. That would be nice as I can put debugging traces near to original traces. I can see that is not easy to implement but I think it is possible.

Thanks.

NourFr10, 2022-10-15:
Hi Tony, how can I start collaborating to the next version of GTKWave ? is there a road map for this purpose? 
Thanks

gtkwave, 2022-10-15:
I have no feature roadmap at the moment.  Primary issue is port to gtk4 and fix what's broken functionwise as there is always something like that with a new toolkit because event loop ordering changes, etc.
Porting from GTK2->3 took me a solid 30 days of all day work.  GTK has non-existent upward compatibility.  I greatly dislike that aspect of the toolkit.

-Tony

Alcame38, 2023-10-11:
Hi,
Does this feature

Alcame38, 2023-10-11:
Hi,

Is this feature still under development ? Have you successfully implemented it ?

Thanks.