gtkwave/gtkwave #155: No rule to make target 'discardbuttons.c', needed by 'discardbuttons.o'

jafinn3, 2022-09-30:
I'm trying to build gtkwave4, and I'm getting this error during `make`. 
```
$ make
<...>
mv -f .deps/currenttime.Tpo .deps/currenttime.Po
gcc -DHAVE_CONFIG_H -I. -I..    -I./.. -I./helpers  -pthread -I/usr/include/gtk-2.0 -I/usr/lib64/gtk-2.0/include -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/pango-1.0 -I/usr/include/fribidi -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/harfbuzz -I/usr/include/freetype2 -I/usr/include/libpng15 -I/usr/include/uuid -I/usr/include/pixman-1 -I/usr/include/libdrm   -I./liblzma -D_WAVE_HAVE_XZ   -I./helpers/fst   -I/usr/include -DHAVE_LIBTCL -I/usr/include       -I./cocoa -pthread -I/usr/include/gtk-unix-print-2.0 -I/usr/include/gtk-2.0 -I/usr/include/atk-1.0 -I/usr/include/cairo -I/usr/include/gdk-pixbuf-2.0 -I/usr/include/pango-1.0 -I/usr/lib64/gtk-2.0/include -I/usr/include/fribidi -I/usr/include/glib-2.0 -I/usr/lib64/glib-2.0/include -I/usr/include/harfbuzz -I/usr/include/freetype2 -I/usr/include/libpng15 -I/usr/include/uuid -I/usr/include/pixman-1 -I/usr/include/libdrm   -DWAVE_GTK_UNIX_PRINT -g -O2  -DFST_WRITER_PARALLEL -MT debug.o -MD -MP -MF .deps/debug.Tpo -c -o debug.o debug.c
mv -f .deps/debug.Tpo .deps/debug.Po
make[4]: *** No rule to make target `discardbuttons.c', needed by `discardbuttons.o'.  Stop.
make[4]: Leaving directory `/home/user/git/gtkwave/gtkwave4/src'
make[3]: *** [all-recursive] Error 1
make[3]: Leaving directory `/home/user/git/gtkwave/gtkwave4/src'
make[2]: *** [all] Error 2
make[2]: Leaving directory `/home/user/git/gtkwave/gtkwave4/src'
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory `/home/user/git/gtkwave/gtkwave4'
make: *** [all] Error 2
<...>
```

Building on CentOS 7

NourFr10, 2022-09-30:
I had this problem too, I believe @rfuest can resolve it.

rfuest, 2022-10-01:
You'll need to use meson to build the `gtkwave4` branch:
```sh
meson build
cd build
meson compile
```

NourFr10, 2022-10-01:
Hi, @rfuest, How can I start collaborating to the GTKWave4 version ?