gtkwave/gtkwave #159: Signal right-click "Open Scope" not working with FST files

wltr, 2022-10-11:
I am used to being able to open the scope of a signal in the tree hierarchy by right-clicking on the signal and then clicking "Open Scope". This used to work with VCD files but somehow doesn't seem to do anything with FST files. (gtkwave v3.3.111)

gtkwave, 2022-10-11:
I wonder if it's broken in the github version (which I don't use)--I don't have time to check that version, but at least the git directory on my PC seems to work.
If it's not the viiewer, something weird is going on with your model.  I use that feature every day.  I just tried in on the des.fst in examples/  Is there some kind of weirdness with escaped hierarchies or something similar?
https://sourceforge.net/projects/gtkwave/files/
...try one of the appropriate versions found here as those are prior to any of the github-specific changes.

-Tony

wltr, 2022-10-13:
Thanks for the reply! It works on `des.fst` with my version too. Looks like that is generated by Icarus. My FST files are generated by Verilator (fstWriter). I'll check in their code.

closed 2022-10-13 by wltr

wltr, 2022-10-13:
I found that it was my use of Verilator + SystemC + FST. After instantiating the DUT within my test module, the top-level name became "test.dut". I assume that "." is used as the scope delimiter. So, in this case, the scope could not be resolved. Instantiating the DUT outside the test class and passing it in, changed the top-level name to just "dut" and everything works as expected.