gtkwave/gtkwave #164: Menu size very small on 4k display

anfractuosity, 2022-11-05:
I'm doing the following which helps increase the font of time scales etc., but not the menu size -

```
gtkwave file.vcd --rcvar 'fontname_signals Monospace 30' --rcvar 'fontname_waves Monospace 30' 
```

I have also tried -

```
export GDK_SCALE=2 
```

But that doesn't seem to have any effect.

Thanks in advance!

gtkwave, 2022-11-05:
It might be a global setting such as here:https://mail.gnome.org/archives/gtk-app-devel-list/2017-May/msg00023.html

...I don't have a 4k display, so I can't directly test this.
-Tony

rfuest, 2022-11-06:
> (I tried with both gtkwave in Arch Linux as well as just building from the gtk3 folder today)

Did you use the `--enable-gtk3` flag when you configured gtkwave? The GTK 2 version doesn't support scaling. The `gtkwave` package in Arch Linux also uses GTK 2.

anfractuosity, 2022-11-06:
Thank you very much both of you!

@rfuest You're right, I hadn't used ```--enable-gtk3```, that did the trick along with ```export GDK_SCALE=2```.

I hadn't realised that the arch package used GTK2 either.

closed 2022-11-06 by anfractuosity