gtkwave/gtkwave #166: Is the gtkwave version of the Ubuntu source not released on GitHub?

lishuai98, 2022-11-15:
Hello，
Why is the speed of loading vcd files different between the gtkwave installed using the official ubuntu source (sudo apt get install gtkwave) and the gtkwave on GitHub? Is the gtkwave version of the Ubuntu source not released on GitHub?

lishuai98, 2022-11-15:
I mean that the speed of loading vcd files after compiling gtkwave released on GitHub is much slower than that of the official Ubuntu source.

rfuest, 2022-11-15:
Did you use the same flags to configure the build? Ubuntu builds use the `--enable-judy` flag.

gtkwave, 2022-11-15:
Again, quantify "much slower" and provide a link to the VCD file if possible.  There should have been no changes that impact perf in the VCD loader.
-Tony

lishuai98, 2022-12-07:
The related compilation environments are as follows.
Ubuntu 22.04.1 LTS 
gcc (Ubuntu 11.3.0-1ubuntu1~22.04) 11.3.0
GTKWave Analyzer v3.4.0 (w)1999-2022 BSI
./autogen.sh
./configure --enable-local-libz
make
make install
Firstly,we used the command "sudo apt-get install gtkwave".(version:GTKWave Analyzer v3.3.103 (w)1999-2019 BSI)
Then we used it to load two vcd files successfully and fastly.But when we used the branch of gtkwave3 on the github,it can't load those vcd files and got stuck.
The related vcd files are shown in the following links: [https://github.com/lishuai98/vcd_test]
[image]

rfuest, 2022-12-07:
Adding `--enable-judy` to the configure call and and installing the libjudy dev package should result in similar performance to the Ubuntu package.