gtkwave/gtkwave #167: meson.build:30:0: ERROR: Dependency "bzip2" not found"

lishuai98, 2022-11-15:
Hello.
I'm trying to use meson to build gtkwave4, and I'm getting this error.

"Run-time dependency bzip2 found: NO (tried pkgconfig and cmake)
 meson.build:30:0: ERROR: Dependency "bzip2" not found"

But I have installed bzip2 on ubuntu, as follows:
bzip2 --version
"bzip2, a block-sorting file compressor.  Version 1.0.8, 13-Jul-2019."

rfuest, 2022-11-15:
You'll need to install `libbz2-dev`.

Markus-Wagner, 2023-04-02:
For me that does not work on Ubuntu 22.04 LTS (libbz2-dev version 1.0.8-5build1). meson version is 1.0.1. I guess, updating build instructions for gtkwave4 would be useful anyway.

Markus-Wagner, 2023-04-02:
Maybe this is helpful: https://bugs.launchpad.net/ubuntu/+source/bzip2/+bug/1939198

Markus-Wagner, 2023-04-02:
I've found a workaround, which solves this particular issue for me: But take it with a grain of salt, as I am not that familiar with pkg-config. The issue is, that some build systems rely on pkg-config to retrieve the correct compiler and linker options for libraries. But the pkg-config tool itselfs relies on `.pc` files, which must be provided by the library packages. In case of the libbz2-dev Ubuntu package, this file is missing (see the bug, I linked in my comment above).
So I crated such a file myself for my particular system configuration, which could be different in your case. The following steps should help you:
1) Figure out, whether it is really the problem. You can try `pkg-config --list-all | grep bzip2`, which should output somethink like this `bzip2                          bzip2 - Lossless, block-sorting data compression`. In case, there is no output, you likely have the same problem, as I had.
2) Get the search paths, pkg-config uses to find .pc files: `pkg-config --variable pc_path pkg-config`.
3) find the directory of the bzip2 shared library, i.e. with `locate libbz2.so`
5) find the directory of the bzip2 header files, i.e. with `locate bzlib.h`
4) Write the pc file using this as a template: https://gitlab.com/bzip2/bzip2/-/blob/master/bzip2.pc.in

In my case, the `bzip2.pc` has the following content:
```
libdir=/usr/lib/x86_64-linux-gnu
includedir=/usr/include

Name: bzip2
Description: Lossless, block-sorting data compression
Version: 1.0.4
Libs: -L${libdir} -lbz2
Cflags: -I${includedir}
```
I omitted some of the fields, which I considered not necessary. Locate will probably yield multiple paths, for the shared library as some programs bring their own version of bzip2. Use the one, which does look, like it does not belong to any other program.
To get the correct version value, I guess the best method is to look, which file libbz2.so points to, as it was a link, not a file on my system.

rfuest, 2023-04-02:
Thanks for looking into this issue. I hope that Ubuntu fixes the bug soon, otherwise I'll add a workaround to the GTKWave 4 build system.

> I guess, updating build instructions for gtkwave4 would be useful anyway.

Yes, the build instructions need to be updated. But at the moment the `gtkwave4` directory doesn't contain many changes and most users/developers should stick with `gtkwave3-gtk3`. I hope to merge a big PR soon(ish), which still doesn't contain many user visible changes but will change the internal architecture and needs lots of testing.

rfuest, 2023-08-22:
This issue has been fixed in PR #234.

closed 2023-08-22 by rfuest