gtkwave/gtkwave #171: Buffer overflows in FST reader

Timmmm, 2022-11-22:
Unless I am missing something, basically every time you read a null terminated string from an FST file you read it into a 512 buffer but don't check that it is less than 512 bytes in length.

[Here's an example](https://github.com/gtkwave/gtkwave/blob/master/gtkwave4/src/helpers/fst/fstapi.c#L4139)

Ok maybe FST reader vulnerabilities aren't that much of a threat in the general scheme of things, but it is a pretty glaring oversight!

gtkwave, 2022-11-23:
There are no real vulnerabilities in most usage cases because people don't pass around FST files like MP3s and generally only view their own generated from a simulator.  Excluding machine generated RTL, I highly doubt anyone would have signal/scope names that long as they would be completely unusable in a tool like gtkwave.

The only real problem most people would encounter is instability/crashes.  However, thanks for pointing it out.  The code has been like that since 2009.

Counting objects: 26, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (15/15), done.
Writing objects: 100% (15/15), 1.46 KiB | 0 bytes/s, done.
Total 15 (delta 14), reused 0 (delta 0)
remote: Resolving deltas: 100% (14/14), completed with 11 local objects.
To https://github.com/gtkwave/gtkwave
   7a0024d..7b97222  master -> master

-Tony

Timmmm, 2022-11-23:
Yeah I agree it is unlikely anyone will ever exploit it! Thought you might want to fix it anyway since it's so easy though and debugging buffer overflows is not fun. Up to you.

gtkwave, 2022-11-24:
The following (cut-and-paste from my previous reply) shows the rev in master where it's been fixed.
-Tony

Counting objects: 26, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (15/15), done.
Writing objects: 100% (15/15), 1.46 KiB | 0 bytes/s, done.
Total 15 (delta 14), reused 0 (delta 0)
remote: Resolving deltas: 100% (14/14), completed with 11 local objects.
To https://github.com/gtkwave/gtkwave
   7a0024d..7b97222  master -> master

Timmmm, 2022-11-24:
Ah I didn't realise you fixed it. Nice one.