gtkwave/gtkwave #173: Performance problems in 3.3.113/gtk3

alexforencich, 2022-11-29:
I recently upgraded from 3.3.111 to 3.3.113 on Arch Linux and I am running in to some seriously bad performance problems in gtkwave.  It almost seems to be effectively hanging when fully zoomed out on some waveform dumps - either scrolling takes like half a second or so per "line", or only the waveform view in the window is rendered.  These issues are not present on 3.3.111, so I have rolled the package back for now.  

Seems like this may be more than a simple version change, though.  The UI changed quite a bit - 3.3.111 has the old window format, while 3.3.113 has the terrible gnome CSD that doubles the size of the title bar and obfuscates the menus.  Looks like 3.3.113 changed the build options to use gtk3, so whatever is going on is probably related to gtk2 vs gtk3.

Changes in the package in question: https://github.com/archlinux/svntogit-community/commit/5bc34f966e038ba00cef0ba59d4c24d7960975b8

alexforencich, 2022-11-29:
Also, here is the specific dump file that I am having trouble with:

[axi_vfifo_raw.fst.zip](https://github.com/gtkwave/gtkwave/files/10108946/axi_vfifo_raw.fst.zip)

I am running gtkwave full screen on a 4k monitor, which probably doesn't help things.  I'm also adding all of the waveforms.  TBH, I don't necessarily mind if the update rate is slow; the problem is the UI being unresponsive to the point where I cannot, say, zoom in on the waveform.

gtkwave, 2022-11-30:
I don't see the problem on my machine.  

You haven't provided much info such as if you installed it from an RPM, etc.
-Tony

alexforencich, 2022-11-30:
It was installed directly from the Arch repo via pacman.  What other information would be useful?

Specific packages in question:

Working: https://archive.archlinux.org/packages/g/gtkwave/gtkwave-3.3.111-1-x86_64.pkg.tar.zst
Poor performance: https://archive.archlinux.org/packages/g/gtkwave/gtkwave-3.3.113-1-x86_64.pkg.tar.zst

Also, here is a screen shot of what happens if I spin the scroll wheel, switch desktops, and switch back:

[image]

Only the actual viewer portion renders, still scrolling one "wave" per second or so.  And htop reports that gtkwave uses a whole CPU core while scrolling.

rfuest, 2022-11-30:
I just tried your file in ArchLinux using the latest GTKWave version in the repository and couldn't reproduce your fault. Are you using X11 or Wayland? I could only test is on X11 at the moment. If you are running Wayland you could try to force the X11 backend by running `GDK_BACKEND=x11 gtkwave`, to see if this improves the performance.

alexforencich, 2022-11-30:
Desktop is XFCE running on X11.  And I just tested with the XFCE compositing setting on and off (it was off before) - that didn't make any difference.

alexforencich, 2023-07-27:
FYI this is still a problem with 3.3.116.  And it doesn't only happen at 4k resolution, it's also repeatable on a 1680x1050 monitor.  On the same machine, 3.3.111 is super smooth.

rfuest, 2023-07-27:
I just tried 3.3.111 from the Arch package and can confirm that it is much smoother than a current version. But the problem doesn't seem to be a simple regression from moving to GTK 3.

Since 3.3.113 the Arch package is build from the `gtkwave3-gtk3` tarball and not the the `gtkwave3` tarball any more. But versions can be built with GTK 2 and there is a big drawing performance difference between the two. The same difference is still present in the latest commit in the `lts` branch.

I haven't done any detailed analysis of the cause of the performance regression, but perhaps the redraw logic, which also causes issues in Wayland, causes to many unnecessary redraws of the waveform.

rfuest, 2023-07-27:
I just realized that the `gtkwave3` version still uses GDK to draw the display and not cairo, like `gtkwave3-gtk3`, which could explain the performance difference.

For GTKWave 4 the drawing model will change significantly, which should at least make vertical scrolling much smoother. At the moment all traces need to be redrawn when the waveform display is scrolled vertically.

I'll take a look at possible optimizations for GTKWave 3 as part of the Wayland fixes. But I don't really want to spend time on more elaborate fixes if the code will need to be changed for GTKWave 4 anyway.

@alexforencich: Just to verify: The performance problems only exist when you have a large waveform window and lots of visible transitions or is performance bad in general?

There is a way to improve drawing speed at the cost of details in areas with lots of transitions. You can increase `WAVE_OPT_SKIP` (https://github.com/gtkwave/gtkwave/blob/master/src/debug.h#L113) to a higher value, but at the moment this is only possible by recompiling GTKWave.

gtkwave, 2023-07-27:
As Ralf says, performance is bad for very dense waves in gtkwave3 branch because XXX_gdk_draw_line uses cairo on each individual line segment instead of wave_gdk_draw_line() bunching up multiple line segments for a single color into a single invocation of gdk_draw_segments().
in comments:
/* completely unnecessary for linux ... unless there are extremely dense traces */
Originally this was for MinGW only on gtk2, but it proved useful in Linux so it's #if 1.
/*
 * gtk_draw_line() acceleration for win32 by doing draw combining
 * now always enabled.  if there need to be an exact correlation
 * between this and the printed data, this needs to be disabled.
 */

...I don't know if cairo allows passing lists of line segments.

-Tony
Message ID: ***@***.***>

rfuest, 2023-07-27:
> ...I don't know if cairo allows passing lists of line segments.

I don't think so. The closest equivalent in cairo would be to have multiple `cairo_move_to` and `cairo_line_to` calls and then a combined call to `cairo_stroke` to draw all segments at once.

In the refactored drawing code I introduced a specialized drawing context for traces, which extends the cairo context with common primitives that are used by the different trace types. I'm not sure how far I got with the code, because I haven't looked at the code in a year, but I think I experimented with adding some optimizations to the drawing code. In very dense areas it should be possible to just draw a single filled rectangle instead of individual transitions.

gtkwave, 2023-07-27:
There are already optimizations that do that at the pixel level so there should be no overdraws.  Cairo apparently is not built to be as fast as X, which doesn't surprise me.
I vaguely remember someone at Novas applying for a patent (this was decades ago so it would be expired) for that rectangle trick you mention.
-Tony
Message ID: ***@***.***>

alexforencich, 2023-07-27:
Seems like it's mainly related to having lots of transitions.  It's bad on both a 4k screen and on a much smaller screen.

I'm also wondering if maybe there are some simple things that could be done to reduce lagginess in extreme cases.  For example, do you really need to redraw on every "click" of the scroll wheel?  Maybe process the shift for each detent, but skip redrawing if there are a bunch of events queued up due to a previous redraw taking far too long.  This way, if the rendering is slow for whatever reason, it won't cause the UI to lock up for an extended period of time, instead it will effectively drop frames to keep up.  

I guess if the new version is going to be released soon, can be a full replacement for version 3, and doesn't have this issue, then it might not be worth wasting time on this.  The Arch package can simply be rolled back to a version that works and held there until version 4 is released.

rfuest, 2023-07-28:
> I'm also wondering if maybe there are some simple things that could be done to reduce lagginess in extreme cases. For example, do you really need to redraw on every "click" of the scroll wheel? [...]

That's the way it should work, but I'm not sure how complicated it would be to hack this behaviour into the current codebase. I will change some of the redraw code to fix issues on Wayland and then spend some time to see if it is possible to improve the drawing performance without huge code changes.

> I guess if the new version is going to be released soon, can be a full replacement for version 3, and doesn't have this issue, then it might not be worth wasting time on this. The Arch package can simply be rolled back to a version that works and held there until version 4 is released.

Unfortunately GTKWave 4 won't be released anytime soon and only 3.4 is planned to be released soonish.

rfuest, 2023-07-29:
I have done some work on improving the drawing performance and at least on my system the performance is much better now. In some instances the drawing time has almost reduced by an order of magnitude. If you want to try it you can compile the PR #224   branch at https://github.com/rfuest/gtkwave/tree/remove-gtk2-master. There is now also a `disable_antialiasing` rc variable to disable antialiasing, which can significantly improve performance.

Sometimes I did also get much better performance in Wayland than in X11, but in the latest version of the PR the difference doesn't seem to be significant anymore.

alexforencich, 2023-07-29:
Just tested it.  Still seems super sluggish.  It takes around 1 second to redraw when scrolling, vs. 3.3.111 which redraws fast enough to not be noticeable.

rfuest, 2023-07-29:
Thanks for testing. 

Could you provide a bit more information about the your testing and your system configuration:
- Did you get the 1 second time with a full screen window at 4K resolution?
- Is there any noticeable improvement with the new branch compared to 3.3.116?
- Does the `disable_antialiasing` rc variable have any effect?
  If you  aren't sure you can measure the drawing time by uncommenting the debug code in this section: https://github.com/rfuest/gtkwave/blob/9202274c393209861ef3f1e834b966d3e860192c/src/wave_view.c#L472-L499
- Which CPU and GPU are you using and which GPU driver?

gtkwave, 2023-07-29:
A GPU should be able to draw millions or tens of millions of lines per second otherwise Spiderman wouldn't be flying around as well as he does on a PS5.

Historically, the biggest time consumption for rendering dense traces across something like 1M+ cycles should be TLB and data cache misses while bsearching histents, but those would be equivalent when rendering GTK2 vs GTK3.  As I said before, there should be no overdraws given how the next x pixel is bsearched for dense traces.  However, I don't know if GTK3 has gone crazy retriggering redraws of the same thing over and over and we don't see that as the redraws are equivalent.  Is something wrong with the frame clock routine in gtkwave accidentally retriggering draws to the backing store when there are no updates needed?

My best guess if it's not that is that Cairo is too top heavy with respect to performance.  If we're really up against a wall, software rendering to a blob of memory that is treated as a copyable pixmap of sorts might be the way to go, but in 2023, as the saying goes, "Really?  You have to be kidding."  I haven't set up timers to measure perf on Cairo calls and maybe I need to do that.  Note that perfwise, GTK2 on X will always be faster as GDK originally was a lightweight abstraction wrapper for X, but I wouldn't expect an order or more of magnitude differences between GDK and Cairo.  That's horrible and is exactly what I saw back when GTK-1.2 was ported to MinGW and the GDK emulation was slow.

For traces that aren't dense, Wayland is noticeably smoother than X.  I see that when using a touchscreen, and flinging traces along with the deceleration that follows is full frame rate and extremely smooth.  It usually blows nWave users' minds.  :-)

-Tony

Message ID: ***@***.***>

rfuest, 2023-07-29:
> A GPU should be able to draw millions or tens of millions of lines per second otherwise Spiderman wouldn't be flying around as well as he does on a PS5.

It should, but I'm not sure how much GPU acceleration is involved in drawing with GTK/Cairo. At some point, while I was working on the PR, I noticed a loud coil whine from my GPU, so there must be at least some GPU involvement. But that was also a time where the performance was even worse that in 3.3.116 :shrug:.

> Is something wrong with the frame clock routine in gtkwave accidentally retriggering draws to the backing store when there are no updates needed?

I don't think it does. If you uncomment the debug commands, I mentioned in my previous post, each redraw of the backing store will print a debug message with the duration in seconds. I didn't notice any unnecessary redraws.

> My best guess if it's not that is that Cairo is too top heavy with respect to performance.  If we're really up against a wall, software rendering to a blob of memory that is treated as a copyable [...]

Cairo sometimes has a bad reputation for not too great performance, but if it is used correctly the performance isn't bad. Unfortunately there doesn't seem to be a lot of documentation about common performance issues or best practices. The main performance improvement in the PR on my system was caused by replacing `cairo_image_surface_create` with `gdk_window_create_similar_image_surface` to create a surface that can be efficiently drawn to the widget.

I would also like to dig into improving the performance more, but would really like to wait until GTKWave 4. My goal for GTKWave 3.4 is to get the performance "good enough". The new drawing code is much simpler and has less code duplication, which will make experimenting with optimizations easier. There will also be one widget per trace with a separate backing image. This should reduce the number of required full redraws, especially for scrolling vertically.

rfuest, 2023-07-30:
I've updated the PR a bit more. The lines are now first stored in a buffer and then drawn all at once using a single call to `cairo_stroke` per color, similar to the buffering implementation in the pre Cairo version of GTKWave. The drawing time is reduced by 15% in one torture test and 65% in another. At the moment this is only implemented for hptr traces and not vptr and anlog traces.

The main contributor to the drawing times are the vector traces, which also benefited the least from the latest optimizations. I haven't had time to investigate this in detail, but disabling the roundcaps has a massive impact. With antialiasing enabled and roundcaps disabled my vector torture test takes ~60 ms to redraw a 2800x1300 pixel widget filled with transitions. With roundcaps enabled the same test takes over 2300 ms. Without antialiasing the difference is much less pronounced ~55 ms vs ~260 ms, which aligns more with the overhead I would expect for the additional drawn lines.

My guess is that horizontal and vertical pixel aligned lines are heavily optimized in Cairo and the slanted antialiased lines are drawn with a much slower code path. If we can change the code to remove the slanted line for the case that a vector is only 1 px wide we should get much better performance for very dense vector traces.

alexforencich, 2023-07-30:
With the latest PR, this is what I am seeing when I open the file, add all waveforms, zoom in one notch, and scroll a bit in all four directions:

```
$ src/gtkwave 

GTKWave Analyzer v3.4.0 (w)1999-2022 BSI

GTKWAVE | Use the -h, --help command line flags to display help.
Draw: 0.000006
Draw: 0.000001

GTKWave Analyzer v3.4.0 (w)1999-2022 BSI

FSTLOAD | Processing 158 facs.
FSTLOAD | Built 158 signals and 0 aliases.
FSTLOAD | Building facility hierarchy tree.
FSTLOAD | Sorting facility hierarchy tree.
Draw: 0.000001
Draw: 0.000003
FSTLOAD | Extracting 158 traces
Draw: 0.001254
Draw: 3.454193
Draw: 3.237412
Draw: 2.980962
Draw: 2.713818
Draw: 3.174916
Draw: 2.976867
Draw: 2.911045
Draw: 2.418227
```

So, it's actually closer to 3 seconds per re-draw now.  And yes, this is full screen on a 4k display.  I did not try changing the antialiasing settings.  CPU is Xeon 6154, GPU is I think a Radeon 5700 XT with the in-kernel driver.

alexforencich, 2023-07-30:
Also, if I roll back that last commit and do the same thing, I get this:

```
$ src/gtkwave 

GTKWave Analyzer v3.4.0 (w)1999-2022 BSI

GTKWAVE | Use the -h, --help command line flags to display help.
Draw: 0.000007

GTKWave Analyzer v3.4.0 (w)1999-2022 BSI

FSTLOAD | Processing 158 facs.
FSTLOAD | Built 158 signals and 0 aliases.
FSTLOAD | Building facility hierarchy tree.
FSTLOAD | Sorting facility hierarchy tree.
Draw: 0.000001
Draw: 0.000001
Draw: 0.000002
FSTLOAD | Extracting 158 traces
Draw: 0.001605
Draw: 1.050485
Draw: 0.916620
Draw: 0.968142
Draw: 0.908251
Draw: 0.841450
Draw: 0.905371
Draw: 0.942445
Draw: 0.858430
Draw: 0.838158
Draw: 0.821878
```

So, seems like that last change had a detrimental effect

rfuest, 2023-07-30:
Hmm, this is strange. I was able to reproduce the reduced performance in the last commit, but only with antialiasing enabled. Without antialiasing the last comment actually improves the performance. I hadn't noticed the issue yesterday because I did most tests with antialiasing disabled.

> I did not try changing the antialiasing settings.

You should try that (`gtkwave --rcvar "disable_antialiasing on"`) and also try to disable roundcaps in the View menu, both should have a huge impact on performance.

rfuest, 2023-07-30:
I've just updated the PR again. The new code reduces overdraw by drawing the entire vector as a closed shape. For small vectors the shape is simplified and for 1 px wide vectors only a vertical line is drawn. This optimization should make very dense areas draw much faster, but some intermediate zoom levels wont be quite as fast because they don't hit the 1 px wide condition as often.

The change causes two small visual difference compared to the previous rendering code:
- The vtrans color is no longer used.
  I hadn't realized that the vector transitions were another color before I worked on that code and I don't think this is a problem.
- Vectors with all bits = 0 are no longer missing the top horizontal line.
  I never liked the way these vectors were drawn, but I can restore this style if necessary. The new code instead draws a thicker bottom line to be consistent with the all bits = 1 vectors.

gtkwave, 2023-07-31:
"Vectors with all bits = 0 are no longer missing the top horizontal line."

  Message ID: ***@***.***>Your call, but where they're handy is to spot all zero values quickly.All 1s, IIRC have a double top bar for the same reason.

That feature was suggested originally to me by Wilson Synder (Verilator/Dinotrace).  At first I didn't like it, but I saw that it was quite useful when looking at waves zoomed out such that no text would be visible, however the all 0s/1s state still could be communicated without having to click down the marker.

-Tony

gtkwave, 2023-07-31:
"How about drawing the all 0s/1s states with reduced opacity for the top/bottom line"Message ID: ***@***.***>
That looks great and it communicates the same information.  I like that idea and implementation.

-Tony

rfuest, 2023-08-01:
The new rendering style and the reduced overdraw has another unexpected advantage. The overdraw did hide the special styling that was applied to all 0s and all 1s states when the transitions were dense. Now more details are visible with medium zoom levels:

Before:
[image]

After:
[image]

The top signal is all 0s most of the time and the bottom signal is all 1s most of the time.

alexforencich, 2023-08-01:
Just ran another test, HUGE improvement this time.  With default settings, I didn't see a draw time higher than 0.25 or so.  With roundcap disabled, looks like it runs closer to 0.05.  But, I think it's easier to read the waveforms with roundcap enabled.  With antialiasing disabled, the results are similar.

But, it does seem to produce some odd artifacts here and there, like so:

[image]

Seems to me like it should go all the way to the bottom on every transition?  This also seems to be pixel position dependent, not value dependent.  If I zoom in/out, different transitions become "short".

Different zoom level of the same thing, these should be totally regular (I think), but there are randomly-placed spikes:

[image]