gtkwave/gtkwave #175: export only selected or visible signals

jotego, 2022-12-03:
The current export feature (in version 3.3.103) exports the whole database. This is akin to a conversion tool to go from one format to another. It would be more useful if you could export only a subset of the database. 

Cadence Simvision allows you to export only the selected signals. Either the selected ones or those visible in the scope will be fine. Exporting a subset of the signals is useful to perform analysis using other tools or to share simulation data with other developers.

gtkwave, 2022-12-03:
I believe that it only exports what signals are available onscreen (including scrolling up/down) or are used for the solid blue and dashed blue marker generation.  This is in order to make a more manageable subset of viewable scope state, for example, from a large FSDB or FST file.

-Tony

jotego, 2024-01-01:
Indeed, that's the workaround I have been using during 2023. It is not ideal, but it is good enough. I will close the issue. Thanks.

closed 2024-01-01 by jotego