gtkwave/gtkwave #179: Is gtkwave4 supposed to build?

jevinskie, 2022-12-18:
I just checked out the repo and tried to build gtkwave4 on Ubuntu 22.04. I receive the following error (and many similar errors for other GTK widgets). Is this expected?

I'd like to help get GTK4 builds working on Mac for A) HiDPI support (which is already pretty good in GTK3) and B) (hopefully) performance improvements (gtkwave with GTK3 on Mac had bad performance regressions compared to GTK2).

I figured I'd start with a sanity check by building for Linux first but, alas, the error.

```
../../git/gtkwave/gtkwave4/src/marker_dialog.h: In function ‘glib_autoptr_clear_GwMarkerDialog’:
/usr/include/glib-2.0/glib/gmacros.h:1195:49: error: ‘glib_autoptr_clear_GtkDialog’ undeclared (first use in this function); did you mean ‘glib_autoptr_clear_GwMarkerDialog’?
 1195 | #define _GLIB_AUTOPTR_CLEAR_FUNC_NAME(TypeName) glib_autoptr_clear_##TypeName
```

rfuest, 2022-12-18:
GTK4 won't build at the moment. While I did get a very rough first version of a GTK4 port working I decided to abandon it and focus on making the GtkWave code more flexible first to make it easier to port in the future. The codebase does, for example, not take full advantage of modern GTK3 features yet, because it still contained GTK2 compatibility until recently, which makes it harder to port to GTK4.

Unfortunately I didn't have much time to work on the `gtkwave4` branch in the last months.

jevinskie, 2022-12-20:
Thanks for the information. For the record, gtkwave4 did build once I used meson and GTK3.

Please let me know if there is some way you think I can help with the cleanup.

closed 2022-12-20 by jevinskie