gtkwave/gtkwave #180: Compiling gtk3 on M1 mac: gtk-mac-integration wrongly picks gtk2

gzz2000, 2022-12-24:
Hi. I just managed to compile gtkwave on an M1 mac following the instructions, but there are bugs that I had to fix myself.

after `brew install gtk+3 gtk-mac-integration tcl-tk xz`, the following steps are needed to compile successfully:
```
./autogen.sh
./configure --enable-gtk3 --prefix=$(pwd)/myinstall --enable-struct-pack "CFLAGS=-I$(brew --prefix)/include -O2 -g" LDFLAGS=-L$(brew --prefix)/lib --no-create --no-recursion --with-tcl=$(brew --prefix)/opt/tcl-tk/lib --with-tk=$(brew --prefix)/opt/tcl-tk/lib
./config.status
grep -Irl gtkmacintegration-gtk2 . | xargs sed -i.bak 's/-lgtkmacintegration-gtk2\ -lgtk-quartz-2.0\ -lgdk-quartz-2.0/-lgtkmacintegration-gtk3/g'
make -j4
make install
```

Note 
(1) the `./config.status` is mandatory, as just running `./configure ...` won't produce `Makefile`. 
(2) all `Makefile`s are linking with gtk3 libraries and **gtkmacintegration-gtk2, gtk-quartz-2.0 gdk-quartz-2.0** simultaneously, which is strange. Without such an ugly `sed` replacement fix, the produced binary complains `GTK+ 2.x symbols detected. Using GTK+ 2.x and GTK+ 3 in the same process is not supported`.

I wonder if there is some bug in the build script.

gtkwave, 2022-12-25:
The bug in the build script is my Mac build machine (an ancient Snow Leopard VM on a laptop) was never updated to gtk3 so I never had an option to test gtk3 so I never made compile options for it and only ever built against gtk2.  Never release what you haven't personally tested...

I had to patch some gdk quartz obj-c code in the official GTK2 sources as well for binaries I produced.  For example, I don't know if the screen grab to file is broken on Mac and produces the wrong screen/colors still, or if it was fixed in GTK2 and/or GTK3.

I don't know how it is now, but the whole process of setting up jhbuild is/was quite involved and as I'm not a Mac user, I never pursued it any further.  If you have working patches, please submit them, thanks.

-Tony

soppelmann, 2023-09-06:
For future reference, the latest release is broken, but it works on 
https://github.com/gtkwave/gtkwave/tree/7b972229b3b674a97702ff63064f714e69b53bb9