gtkwave/gtkwave #181: Hex data format with X bits

rfuest, 2023-01-18:
I'm currently refactoring the code which converts `HistEnt` vectors to strings and have a question about the intended behavior if a vector includes `X` values:

The code does use the `xover` variables to give `X` values precedence over other values. The nibble `00x0` would, for example, be displayed as `X` in the hex format. But if multiple special values are mixed only the first different bit in the nibble is considered: `wx00` is displayed as `X`, but `w0x0` is displayed as `W`. Is this behavior intentional or should both cases be displayed as `X`?

gtkwave, 2023-01-19:
That's a bug.  I never use VHDL anymore (and even back in the day it was only IBM two-state sim), so I haven't encountered it.  

I'd say X gets precedence over everything, mainly because multiple drivers with different values (traditional definition for "x", not "x poisoning" on non-valid qualified signals on an interface)  is almost certainly unintended behavior that needs a beacon shown on it.

-Tony

rfuest, 2023-01-19:
OK, thanks for confirming my assumption that this is a bug. I'll change the behavior in the `gtkwave4` branch and also modify the octal display format to work the same way. At the moment the octal format just uses the first special value and it doesn't change the between lower and upper case to indicate whether all bits, that belong to one digit, are the same special value.

closed 2023-02-28 by rfuest