gtkwave/gtkwave #193: Error when opening FST

shareefj, 2023-03-24:
Hi,

having just compiled the latest `gtkwave3` I get the following error when loading an FST.  The same FST loads fine in the Gtkwave I'm using via `apt install gtkwave` (v3.3.104).

```
GTKWave Analyzer v3.4.0 (w)1999-2022 BSI

FSTLOAD | Processing 99772 facs.
FSTLOAD | Internal error, nonsequential enum tables definition encountered, exiting.
```

gtkwave, 2023-03-24:
What's this FST from?  nvc?
I haven't seen that error before.  There is code above it that is ifdef conditional based on Judy arrays (_WAVE_HAVE_JUDY), so I don't know if support without Judy is broken.  I doubt you're compiling against Judy if you're going from source, but you'll want to try the opposite of what you're doing.

-Tony

deo248, 2023-03-27:
I also just encountered this error. My FST was produced by verilator 5.006 and the version of gtkwave is 3.3.114. My ./configure summary is attached - the only option I specified was --prefix, all other options left as default. The same FST works with v3.3.111, which was also compiled from source similarly.

[configure.txt](https://github.com/gtkwave/gtkwave/files/11082934/configure.txt)

deo248, 2023-03-27:
After recompiling and reinstalling with Judy enabled (and installing libjudy-dev for Ubuntu 18.04), I have the same issue with the same error message. Updated configuration summary attached.

[configure_judy.txt](https://github.com/gtkwave/gtkwave/files/11083036/configure_judy.txt)

deo248, 2023-03-27:
I did a little more poking around and gtkwave version 3.3.113, both with and without judy enabled, successfully opened the same FST.

shareefj, 2023-03-28:
Sorry, missed these comments.

Yes, it was from Verilator but v4.224.

rfuest, 2023-03-28:
Do you encounter this problem for all FST files generated by Verilator? I did recently used Verilator with GTKWave 3.3.114 and didn't notice any problems.

Can you share a FST file, which causes the error?

gtkwave, 2023-03-28:
3.3.114    23nov22    Buffer overflow fixes in FST reader.

...fix a reported "security" problem and break function.  Please attach an FST file that breaks so I can see what has broken.

-Tony

deo248, 2023-03-28:
Unfortunately I cannot supply the FST that triggers the problem on my end.

I went and pulled the picorv32 source in the hopes of recreating something open-source there and ran an FST trace but the produced file did not trigger the error.

I did a little bit of investigation and was able to somewhat localize the source code that triggers the problem on my end down to 3 of my source RTL files, all SystemVerilog. If that is a productive avenue of approach to help in lieu of providing an FST, then I may be able to continue there and narrow down some syntax that produces something in an FST that triggers the problem.

gtkwave, 2023-03-28:
#define FST_ID_NAM_SIZ                  (512)

...I'd say increase the value of this to something very large like 4096 to see if Verilator is emitting extremely long signal names, possibly flattened or something like that, and we never noticed before.  Diffing between 3.3.113 and 3.3.114 doesn't show a lot of places this can be breaking.

-Tony

deo248, 2023-03-28:
Changing `FST_ID_NAM_SIZ` from `(512)` to `(4096)` in `fstapi.c` does in fact solve the problem for me.

Just for additional information, 1024 did _not_ work in my test, but 2048 and above do work.

gtkwave, 2023-03-28:
It sounds to me like verilator is emitting some super long signals/attributes or something like it wrote out a string missing a null termination.  You might have to ask Wilson Snyder.  Alternatively, run fst2vcd on the fst file ensuring you have the -e/--extensions flag set and see if there are any super long lines potentially runaway string lines that are $ directives.
-Tony

deo248, 2023-03-28:
The longest single line in the processed VCD is a 1452 character line. It's an `$attrbegin misc 07` line for an enum. That enum has 33 members which are implicitly assigned values, meaning they range from integer 0 to integer 32, each of which is then represented in binary in the attribute line, which makes for a very long attribute line.

I have included a lightly edited version of this line below. The original is 1452 characters in length - this one is 1394 - but it serves to illustrate the possible culprit.

```
$attrbegin misc 07 XXXXXX.__typeimpenum89 33 STATE0 STATE1 STATE2 STATE3 STATE4 STATE5 STATE6 STATE7 STATE8 STATE9 STATE10 STATE11 STATE12 STATE13 STATE14 STATE15 STATE16 STATE17 STATE18 STATE19 STATE20 STATE21 STATE22 STATE23 STATE24 STATE25 STATE26 STATE27 STATE28 STATE29 STATE30 STATE31 STATE32 00000000000000000000000000000000 00000000000000000000000000000001 00000000000000000000000000000010 00000000000000000000000000000011 00000000000000000000000000000100 00000000000000000000000000000101 00000000000000000000000000000110 00000000000000000000000000000111 00000000000000000000000000001000 00000000000000000000000000001001 00000000000000000000000000001010 00000000000000000000000000001011 00000000000000000000000000001100 00000000000000000000000000001101 00000000000000000000000000001110 00000000000000000000000000001111 00000000000000000000000000010000 00000000000000000000000000010001 00000000000000000000000000010010 00000000000000000000000000010011 00000000000000000000000000010100 00000000000000000000000000010101 00000000000000000000000000010110 00000000000000000000000000010111 00000000000000000000000000011000 00000000000000000000000000011001 00000000000000000000000000011010 00000000000000000000000000011011 00000000000000000000000000011100 00000000000000000000000000011101 00000000000000000000000000011110 00000000000000000000000000011111 00000000000000000000000000100000 6 $end
```

gtkwave, 2023-03-29:
OK, that's the area of what is causing the problem.  I also noticed off by one errors on the string length checks...

Try grabbing the fstapi.c from either sourceforge or github.  If you're going to be using 3.3.114, sourceforge will have the 3.3.115 maintenance release soon and this update will be in it.
Thanks,-Tony

gtkwave, 2023-03-30:
> I can confirm that 3.3.114 compiled with the updated fstapi.c file does successfully open the same FST I've been using as a test. I appreciate the quick help!

Thanks for verifying that it works.  I'll kick out a 3.3.115 bugfix maintenance release sometime this weekend that incorporates it.

-Tony