gtkwave/gtkwave #196: How to use wildcards in string pattern search

shareefj, 2023-04-11:
I'm trying to work out the correct syntax for searching for wildcard signal values.  If I got to `Search->Pattern Search 1` and select `String` from the dropdown menu, I was expecting to be able to enter a wildcard expression such as `CAFE000?`.  Is this not possible?  It's such basic functionality that I'm hoping I'm missing something obvious.

gtkwave, 2023-04-11:
No such function exists.  If I had to search for such a pattern, generally, I leave off that "?" character.  The search looks for the string anywhere in the value: it is not a 1:1 direct match.
-Tony

shareefj, 2023-04-13:
Ah okay, I hadn't thought of that but it still doesn't work for matching a wildcard in the middle of the string.  Is this something that could be implemented in the future or is it a non-starter?

gtkwave, 2023-04-13:
Sure, it could be implemented in the future.Replace strstr_i() with a compiled regex compared and problem solved.See, for example wave_regex_compile and wave_regex_match in regex.c.

I just don't have the time to work on it based on where my day job's at in the project, but that's where to do it.
-Tony