gtkwave/gtkwave #197: Bug in text box for pattern search

shareefj, 2023-04-11:
Hi,

there's an annoying bug in the text box within the pattern search window.  If you enter a value and then go back to edit it, deleting any character then seems to auto highlight the whole string which is overwritten with further typing.  So to modify a string you first have to delete a single character at a time followed by pressing the right arrow key to un-highlight the string.  PITA.

This is on 3.3.104 and 3.3.115 which is all I've tested.

Shareef.

gtkwave, 2023-04-11:
The annoying bug is because most people won't press enter to update search results before pressing one of the buttons, so the question is how to get the current string data reported by the widget.  I don't know if it would require a custom widget or a different method to update the current string data for searches.

-Tony

shareefj, 2023-04-13:
OK, so just for my understanding, you have a callback that fakes an enter key press because the new value needs to be updated before any button is pressed?

And will the suggested fix by @rfuest be made?

gtkwave, 2023-04-13:
> I don't see why it should be changing the selection. It might've been a holdover from the GTK 1.2 days that's no longer required.
>
> I did some quick research using the gtkwave-archive repo (primarily to test the usefulness of the archive). You are correct that this code is very old and did exist at the beginning of the CVS repo. In this commit from 2007 the enter_callback was then also connected to the changed signal: ***@***.***#diff-b3dbb9647cce4420b8d5073ed1c402c93be4d3001bc966dc47dc712ab88fe38d
>
> In the past there was probably a good reason to change the selection on pressing enter, but the selection shouldn't have been modified on all changes to the text. For modern code I would suggest to keep manual changes to the selection to a minimum and to rely on the default behavior of the toolkit.

Sure, totally, include the change.
-Tony

I can prepare a PR for this change if @gtkwave likes to include this change.

closed 2023-04-17 by gtkwave