gtkwave/gtkwave #200: Transaction Filter Process feature not working in macOS

jiegec, 2023-04-30:
Hi, I am using gtkwave-gtk3 3.3.115. If I use Transaction Filter Process feature of gtkwave in macOS, gtkwave will hang and allocate memory in a loop. Digging into the code, the culprit is that the subprocess exited and feof() returnes failure:

```c
				if(feof(GLOBALS->ttrans_filter[t->t_filter]->sin)) break; /* should never happen */
```

The subprocess was exited due to multithreaded-fork error:

```
Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BREAKPOINT (SIGTRAP)
Exception Codes:       0x0000000000000001, 0x00000001a2ea9238

Termination Reason:    Namespace SIGNAL, Code 5 Trace/BPT trap: 5
Terminating Process:   exc handler [19784]

Application Specific Information:
*** multi-threaded process forked ***
BUG IN CLIENT OF LIBPLATFORM: os_unfair_lock is corrupt
Abort Cause 258
crashed on child side of fork pre-exec

Kernel Triage:
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage

Thread 0 Crashed::  Dispatch queue: com.apple.main-thread
0   libsystem_platform.dylib      	       0x1a2ea9238 _os_unfair_lock_corruption_abort + 88
1   libsystem_platform.dylib      	       0x1a2ea47e8 _os_unfair_lock_lock_slow + 332
2   ???                           	       0x10622bed8 AtForkPrepare + 44
3   libsystem_pthread.dylib       	       0x1a2e77680 _pthread_atfork_prepare_handlers + 96
4   libSystem.B.dylib             	       0x1ae7e84e8 libSystem_atfork_prepare + 32
5   libsystem_c.dylib             	       0x1a2d318f8 fork + 36
6   ???                           	       0x104cfa120 pipeio_create + 356 (pipeio.c:149)
7   ???                           	       0x104d2b2ac load_ttrans_filter + 332 (ttranslate.c:211)
8   ???                           	       0x104d2b3ac add_filter_callback_2 + 168 (ttranslate.c:381)
9   ???                           	       0x104cccc4c fileselbox + 348
10  ???                           	       0x104d2af9c add_filter_callback + 96 (ttranslate.c:413)
11  ???                           	       0x105148ab0 g_closure_invoke + 200
12  ???                           	       0x105160914 signal_emit_unlocked_R + 1864
13  ???                           	       0x1051616a8 g_signal_emit_valist + 2556
14  ???                           	       0x1051619bc g_signal_emit + 44
15  ???                           	       0x10569dd94 gtk_button_clicked + 132
```

The first fork from gtkwave to its child succeeded. The fork in child process crashed. If I remove the code for monitor process, it just works:

```c
#ifdef _AIX
	/* NOTE: doesn't handle ctrl-c or killing, but I don't want to mess with this right now for AIX */
	if ((!arg)||(strlen(arg) == 0)) {
	  execl(execappname, execappname, NULL);
	} else {
	  execl(execappname, execappname, arg, NULL);
	}
	exit(0);
#else
		if (strlen(arg) == 0) {
		  execl(execappname, execappname, NULL);
		} else {
		  execl(execappname, execappname, arg, NULL);
		}
		perror("execl");
exit(0);
```

Essentially duplicating the AIX code. Possible solutions:

1. Disable monitor process for macOS
2. Find another way to avoid leaking subprocess

gtkwave, 2023-04-30:
Why are you writing processes that exit without emitting $finish first?  :-)

From a gtkwave code robustness point of view, if the subprocess exited, would
if(feof(GLOBALS->ttrans_filter[t->t_filter]->sin)) {is_finish=1; break;} /* should never happen */

be more correct as it would implicitly create a '$finish' command and break the overarching while(!is_finish) loop?  Please try that for me.

-Tony

jiegec, 2023-05-01:
Nope, the process does not exit by itself. It violates the forking check of multithreaded process in macOS and exits before exec()-ing to the actual transaction filter process.

gtkwave, 2023-05-01:
What version of MacOS is this?  I'm wondering how something so fundamental has changed in code that has run properly since Snow Leopard.  It's possible there is a race condition in the C code which is why AIX has a special ifdef, but I haven't dealt with AIX at least since around 2012 when I worked at IBM.

I don't have a current Mac so I can't test any of this and other users would have to test and/or comment.

-Tony

jiegec, 2023-05-02:
macOS Ventura 13.3.1

niklaut, 2023-05-23:
Yup, can confirm, I see the same behavior on macOS 13.3.1.