gtkwave/gtkwave #202: [feature] Shell completions

Freed-Wu, 2023-05-17:
Can this project provide completions for common shells? TIA!

gtkwave, 2023-05-17:
I don't know what you mean by that question.  Do you have an example of what you mean?
Thanks,-Tony

Freed-Wu, 2023-05-17:
Like <https://github.com/steveicarus/iverilog/issues/921>

gtkwave, 2023-05-17:
Unfortunately, that still says nothing to me.  How with respect to using gtkwave would (what I presume) zsh style shell completion be used?  In the Tcl/wish command line?  What?

Thanks,
-Tony

 
 

Like steveicarus/iverilog#921

Freed-Wu, 2023-05-17:
> How ... shell completion be used?

```shell
$ gtkwave -<TAB>
option
-1               specify RPCID of GConf session
-2               specify new current working directory
-3               restore previous session
-4               specify single rc variable values individually
-5               specify sst exclusion filter filename
-6               set gtk-application-prefer-dark-theme = TRUE
-7               prompt user to write save file at exit
-A               assume savefile is suffix modified dumpfile name
-a               specify savefile name
--autosavename   assume savefile is suffix modified dumpfile name
--chdir          specify new current working directory
--comphier       use compressed hierarchy names (slower)
--cpu            specify number of CPUs for parallelizable ops
-c               specify number of CPUs for parallelizable ops
-C               use compressed hierarchy names (slower)
--dark           set gtk-application-prefer-dark-theme = TRUE
--defaultskip    if missing .rcfile, do not use useful defaults
-d               if missing .rcfile, do not use useful defaults
...
```

gtkwave, 2023-05-17:
Please be descriptive.  Use Chinese to express yourself if you need to: I studied it in college.
-Tony

Freed-Wu, 2023-05-18:
Thanks. Users can use this shell completion by press `gtkwave -<TAB>` to get the supported options or `gtkwave <TAB>` to get the possible `*.vcd`. That is how the shell completion can be used.