gtkwave/gtkwave #203: sprintf to be depreciated for macOS

allexoll, 2023-05-26:
sprintf is depreciated on macOS, and it raises a lot of warnings for project using the fst helpers like verilator.

```sh
/Library/Developer/CommandLineTools/SDKs/MacOSX13.sdk/usr/include/stdio.h:188:1: note: 'sprintf' has been explicitly marked deprecated here
__deprecated_msg("This function is provided for compatibility reasons only.  Due to security concerns inherent in the design of sprintf(3), it is highly recommended that you use snprintf(3) instead.")

```
I raised an [issue there](https://github.com/verilator/verilator/issues/4141), and they told me it made more sense to fix it upstream (here), so I'm raising the issue here as well. 

I can give it a go, but I'm unclear what the process would be for this repository (multiple copies of the code), and what the validation process is (if any)

rfuest, 2023-05-26:
Verilator does only include the FST library from the GTKWave repository. If you want to fix the issue you'll need to change all three copies of the code in the `src/helpers/fst` directories.

@gtkwave would you be interested in creating a separate repository for the FST library? IMO this would make it easier to add a dependency to the FST library from other projects.

gtkwave, 2023-05-26:
I'll look at the helpers/ directory source files in the near future.  As far the Verilator side of things go, there aren't any real security implications for FST files: it's not like someone is going to be trading waves like mp3s back in the day on Napster.  Moreover, many of those sprintf usages already malloc the number of bytes needed (or more than needed) just ahead of time of usage in the same function, so they're good.  IIRC, static code analysis using some LLVM bolt-on tool identified many problem areas years ago.
If Apple pulls the rug quickly, a wrapper define can easily make things compilable again.  My only real concern with sweeping updates of the sort is "let's fix potential security problems" breaking code due to introduced typos or whatever.  There are probably ~300 usages across the entire viewer codebase which isn't a crazy amount, but it's large enough to require some amount of concentrated effort.

-Tony

gtkwave, 2023-05-26:
The initial reason for putting gtkwave up on github was so Wilson could integrate into Verilator more easily, however that process worked for him to do that.  As far as I know, the only high-visibility consumers of the code outside of gtkwave are Verilator, Icarus, nvc, and CVC.  I know there are other users, but I can't think of any other fairly well-known or large projects in the open source EDA world that use it.  Nobody's forked it yet to do that, for example.

If a separate repo works, I'm not against it, and it would cut down on what needs to be updated.  I'll let it be your judgment call.  BTW, I still need to turn over github gtkwave project ownership over to some other person or collective entity like you mentioned a while back.  I don't have that note handy on what was discussed as I was drowning in day job deadlines at that time.  I still am drowning, but not so badly.  :-)

-Tony

gtkwave, 2023-05-27:
> Counting objects: 26, done.
> Delta compression using up to 4 threads.
> Compressing objects: 100% (15/15), done.
> Writing objects: 100% (15/15), 1.61 KiB | 0 bytes/s, done.
> Total 15 (delta 14), reused 0 (delta 0)
> remote: Resolving deltas: 100% (14/14), completed with 11 local objects.
> To https://github.com/gtkwave/gtkwave
>    cf84afb..048a5c1  master -> master

I just updated the fstapi files, mainly because they have visibility outside the immediate gtkwave project.  I'll look at the helpers utils when I have the time.

-Tony

638:/home/bybell/git/gtkwave> git push origin masterUsername for 'https://github.com': gtkwave

rfuest, 2023-05-29:
> If a separate repo works, I'm not against it, and it would cut down on what needs to be updated.  I'll let it be your judgment call.  BTW, I still need to turn over github gtkwave project ownership [...]

I think it would be nice to have another repo for the FST lib in the long term, but it's not something I want to work on at the moment. If I let myself get distracted by this change I will never finish the first part of the GTKWave 4 refactor, which I didn't make much progress on recently because life did get in the way.

Regarding the use of `sprintf` in the codebase: I've already changed quite a few instances of `sprintf` to the string functions provided by `glib`, like `g_strdup_printf` or by using `GString`. This does not only reduce the risk of buffer overflows, but also makes the code more readable.