gtkwave/gtkwave #208: Variables in the root scope can't be accessed using SST

rfuest, 2023-06-15:
If a dump file contains variables in the root scope it is not possible to access them using the SST after another scope has been selected.

Example VCD file:
```vcd
$var wire 1 a root_var $end
$scope module mod $end
$var wire 1 a other_var $end
$upscope $end
$enddefinitions $en

#0
a0
```

After GTKWave is started the variables in the root scope are displayed:
[image]

But as soon as another scope is selected it is no longer possible to access the variables in the root scope via the SST. They can only be added using other methods, like the Signal Search Hierarchy.
[image]

The only solution to this issue I could come up with is to add a root scope node to the SST, similar to the `top` node which is added for GHW files.

There are two remaining questions if we decide to add a root node to the SST:
- How should the node for the root  scope be called?
  Using `top` isn't ideal IMO, because `top` is a common name in HDL hierarchies and this could lead to situations where two nested `top` nodes are present. I think `Root` would be better, but I'm open to better suggestions.
- Should the root node only be added if necessary or always?
  Adding a root node is only necessary if there are variables in the root scope, which isn't the case for many dump files. I usually prefer a consistent behavior for all dump files, but not adding the root scope if it isn't necessary would reduce the nesting depth in the SST.

gtkwave, 2023-06-15:
Shift-click the root scope to deselect it and your toplevel signals should come back.
-Tony

rfuest, 2023-06-16:
OK, CTRL-click to deselect does work. But this is hard to discover for users and I think we should make it easier to access these signals.