gtkwave/gtkwave #211: Multidimensional arrays in Verilog converted to Single dimensional

AhsanAliUet, 2023-06-19:
Hi,
I am simulating my verilog code on VCS simulator. I have a signal which is ```reg [2:0] sig_0 [3:0];```, when I see this signal in GTKwave, the signal is converted to single dimensional array of size [11:0] which is [(2+1)*(3+1)-1:0] which is of size 12. The data in is same but it is much difficult to see it by ourselves.

Also, same thing happens for struct in verilog which becomes headache if converted to a single dimensional array.

Do we have solution for this in gtkwave?

Thansk in advance.

umarcor, 2023-06-19:
Did you check the content of the waveform file generated by VCS? Maybe the conversion is done by the simulator, so GtkWave cannot know that they are coming from multi-dimensional arrays or structs.

gtkwave, 2023-06-19:
There is no solution.  Simulators such as Icarus will make a problem out of some signals such as these.  I don't remember if it's for this case or packed arrays.  VCS and other simulators don't exhibit this.

-Tony

closed 2023-07-08 by AhsanAliUet