gtkwave/gtkwave #212: gtkwave crashes after trying to reload a deleted trace

LiamSkirrow, 2023-06-22:
My simulation flow involves deleting and recreating the directory containing my waveforms from scratch. It seems that gtkwave doesn't like this approach as when I reload the waveform (ctrl+shift+R) it just hangs for a while and I have to force quit the program.

I'm running gtkwave v3.3.61 on Ubuntu 20.04.6 LTS. I'm happy to take care of this in a PR, just let me know roughly where I should be looking in the source code :+1:

rfuest, 2023-06-22:
Did you check if the issue still exists in newer versions of GTKWave? 3.3.61 is 9 years old. You could, for example, try to run the Flatpak version.

LiamSkirrow, 2023-06-22:
Fair enough, I'll run it on a newer version and will report back. Thanks.

rfuest, 2023-06-22:
The FST error has been fixed in release 3.3.115, but apparently the Flatpak version hasn't been updated yet.

LiamSkirrow, 2023-06-23:
Just to add another datapoint, I tried running gtkwave v3.3.103 and I still have to force close the program when reloading on a deleted trace. 

I don't have immediate access to a more up to date version currently but happy to keep investigating. Where can I actually download the most up to date version of gtkwave3? Shall I build source from the `gtkwave3` dir in this repo to get the latest version? Thanks.

LiamSkirrow, 2023-06-23:
Ok I built from source (gtkwave3) to run v3.4.0 (Copyright message suggests this dates from 2022 so is probably the most recent version) and can confirm the bug is still persisting, exactly the same with the other versions. 

I'll get working on a fix for this and will submit a PR soon enough :sunglasses:

rfuest, 2023-06-23:
The `gtkwave3-gtk3` contains the current GTK 3 version. Be sure to use the `--enable-gtk3` flag when you call `./configure`. There are some build instructions in the readme file at https://github.com/gtkwave/gtkwave/tree/master/gtkwave3-gtk3.

rfuest, 2023-06-23:
I wasn't able to reproduce the error. Could you provide a set of files and a step by step instruction which triggers the bug?

LiamSkirrow, 2023-06-23:
I was able to reproduce the error by building `gtkwave3-gtk3` following the instructions in the readme.

To reproduce:
```
- cd ./dir-containing-trace/
- gtkwave test.fst
- mkdir tmp && mv test.fst ./tmp    # move the trace to a different area, alternatively just delete it
- [reload waveform -> ctrl+shift+R] 
- *gtkwave freezes and doesn't recover, have to kill process*
```

gtkwave, 2023-06-23:
Of course it freezes.  Read the stdout text.  It's waiting for a file to appear and you trashed it so it goes into a wait loop.  You have no sim file.
-Tony

LiamSkirrow, 2023-06-23:
Yes but I feel this should be handled gracefully and not cause the program to freeze, warranting having to manually kill the process.

Even when the trace file is reproduced in the same location, reloading the waveform still freezes everything. Semantically, I'd label this as a bug. Again, happy to come up with a fix, but only if I have support that it'll make its way into master.

rfuest, 2023-06-23:
> Even when the trace file is reproduced in the same location, reloading the waveform still freezes everything. Semantically, I'd label this as a bug. Again, happy to come up with a fix, but only if I have support that it'll make its way into master.

Does this also happen if you try to reload after the trace file has already been replaced?

LiamSkirrow, 2023-06-23:
I'll give #213 a go :+1: 

> Does this also happen if you try to reload after the trace file has already been replaced?

Yes, although under the condition that you run gtkwave in the directory that gets deleted. 

So to elaborate a little, I dump a trace to a given directory, and invoke gtkwave from that dir and then proceed to rerun an RTL simulation (which involves blowing away that directory and recreating it again, dumping the new trace file there) At this point if I reload the trace file, gtkwave will freeze.

gtkwave, 2023-06-23:
File overwrites such as rerunning sim are ok.Deleting the dumpfile causes the viewer to wait for the file to reappear.Deleting the directory is fatal.
-Tony

 
 

I'll give #213 a go 👍

rfuest, 2023-06-23:
#213 wasn't meant to change the behavior and was only intended as a quick hack to at least not get any warning by the desktop environment that the window is unresponsive. I agree with you that immediately failing and showing an error in the UI would be a better default behavior for the reload function. The code that loads files isn't something I would like to touch in GTKWave 3, because it is rather complicated, but I think it would be a good idea for GTKWave 4, if Tony agrees.

gtkwave, 2023-06-23:
I'd say leave GTKWave 3 alone.
The best interim fix I can think of would be to see if the reload file exists (can be fopened or gets good stat(2)), then reload is refused before the context destroy starts and control is returned to the viewer.  Note that once the viewer context is destroyed, you've lost everything except for a temporary save file which exists in your tempfile directory and will appear similar to this:
-rw-------.  1 bybell bybell      547 Jun 23 11:46 gtkwaveZUEvLh

-Tony

LiamSkirrow, 2023-07-04:
Just to add a suggestion for gtkwave4 on this same topic:

When gtkwave tries to read a trace file (that's been deleted) it sits in a loop waiting for it to magically reappear. What's the purpose of this loop? It's the loop that causes the lack of responsiveness (giving the frozen window), probably due to the `sleep()` function, since I suspect that disallows the execution of any callbacks etc while it's running.

To my mind it makes more sense to issue an error message in the terminal and do nothing else, simply returning to the main loop and not trying anything else, this would eliminate the window freezing and I reckon it's more sensible functionality in the long run.

Thoughts on making this 'the way' for `gtkwave4`?

rfuest, 2023-07-04:
The current code makes heavy use of a shared global struct (`GLOBALS`), which is read any modified in many places. This makes it hard to decouple different parts of the code and I guess the reload code is blocking to make sure no other parts of the code interfere with the loading process.

In GTKWave 4 the savefiles and tracefiles can be loaded without accessing any shared global data. This way all loading can be moved into a separate thread, which won't interfere with the UI updates. Once the loading is finished (or has failed) the main thread will be notified to update the UI.

closed 2023-07-04 by LiamSkirrow

gtkwave, 2023-07-04:
"What's the purpose of this loop?"
The purpose of that loop was to deal with reload failures when someone reran sim and clicked reload but the header data for rerun simulation wasn't yet written out to disk and the load would fail, for example, due to AFS or LSF/LoadLeveller latency.  Wait with backoff, then eventually the file exists.  I never deleted wave directories in my env: they were always located in a well-known position, etc.  Different usage cases and that loop doesn't work how you use the viewer.
-Tony

punzik, 2024-02-28:
Maybe add an option to select gtkwave behavior in the absence of a file? I think that would be great.