gtkwave/gtkwave #214: Saving user preferences

OpusElectronics, 2023-06-24:
gtkwave doesn't seem to save user preferences, unless I've missed it.
For instance, it would be nice if the View and Search options (pretty much anything that can be set with a checkbox) were saved as preferences and restored when gtkwave is launched.

LiamSkirrow, 2023-06-24:
You can do this by saving a `.gtkwaverc` file to your home directory. Type `man .gtkwaverc` for a list of 'settable' options.

The maintainers mentioned this in an old issue or PR but I can't seem to find it right now.

OpusElectronics, 2023-06-24:
Thanks! That's apparently 'man gtkwaverc' without the dot.
I'll have a look at this. Should do for now.

Still,  I think it would be a reasonable idea to automatically save it by default from the current app settings.
There may be a reason (that I don't know) why they decided not to do this.

OpusElectronics, 2023-06-24:
Small problem, I get a segmentation fault when launching gtkwave, when a .gtkwaverc file is present in my home directory.

Note that I only added the options I wanted to be different from default in this file.
Is it what makes it crash? Should the config files list absolutely all known options to avoid the crash?
Could you test on your side, with a .gtkwaverc with only one option in it?

OpusElectronics, 2023-06-24:
Minimal example that makes it crash, create the following .gtkwaverc file in your home directory:
```
show_base_symbols yes
```

LiamSkirrow, 2023-06-25:
I believe that should be 
```
show_base_symbols 1
```

So that's a '1' instead of a 'yes'. 

> Should the config files list absolutely all known options to avoid the crash?

Nah you only need to include the options that are different from the default values, for example my .gtkwaverc file has only five settings.

Which version of gtkwave are you running?

rfuest, 2023-06-25:
I couldn't reproduce the segfault with the GIT version of GTKWave and `gtkwave --rcvar "show_base_symbols 1" examples/des.gtkw` works as expected. But with a binary that is compiled from the SourceForge tarball the same command causes a segfault. Some fix in the GIT version must not have been applied to the SF version.

OpusElectronics, 2023-06-25:
> I believe that should be
> 
> ```
> show_base_symbols 1
> ```

I had tried it too. Doesn't make a difference, it does crash as well.
(Note that I based the 'yes' on the example config file that can be found for instance here:
https://github.com/gtkwave/gtkwave/blob/master/gtkwave3-gtk3/examples/gtkwaverc
The same content can be found there:
https://github.com/gtkwave/gtkwave/blob/master/gtkwave4/examples/gtkwaverc
I'm not sure how up-to-date this example is. Maybe examples should get updated? Or removed?)

> So that's a '1' instead of a 'yes'.

Again doesn't seem to make any difference.

> > Should the config files list absolutely all known options to avoid the crash?
> 
> Nah you only need to include the options that are different from the default values, for example my .gtkwaverc file has only five settings.

Thought so, but was just asking in case that could have been the cause of the crash.

> Which version of gtkwave are you running?

3.3.115, which is the last "release" as far as I can tell. At least that's what we can figure out from Sourceforge, and I guess that's what many Linux distros base their packages on.
I'm using the official gtkwave package on Arch.

If this crash is fixed in the current git version, that's all good. But maybe this issue may be an opportunity to more clearly define what releases of gtkwave are? 3.3.115 is the latest release that can be downloaded from Sourceforge, and that is mentioned on its web page: https://gtkwave.sourceforge.net/ , but OTOH, there doesn't seem to be any release tagged on the github repo, apart from a 'nightly' version, which is not really a release.

Just mentioning this because again most Linux distros with a gtkwave package usually base it on the Sourceforge info and source tarballs.

gtkwave, 2023-06-25:
I don't see the problem at all either with specifying the var on the command line or using a .gtkwaverc file.

Note that '1' vs 'yes' vs 'on' are synonyms unless that function was broken somewhere.
I don't use Arch so I can't test how this was built.  I just built from source on RHEL7.

-Tony
581:/tmp/gtkwave-3.3.115/src> gtkwave --rcvar "show_base_symbols 1" ../examples/des.gtkw 

GTKWave Analyzer v3.3.115 (w)1999-2023 BSI

RCVAR   | 'show_base_symbols 1' FOUND
FSTLOAD | Processing 1432 facs.
FSTLOAD | Built 1287 signals and 145 aliases.
FSTLOAD | Building facility hierarchy tree.
FSTLOAD | Sorting facility hierarchy tree.

###
/*
 * atoi 64-bit version..
 * y/on     default to '1'
 * n/nonnum default to '0'
 */
TimeType atoi_64(const char *str)
{
...

 
 

https://github.com/gtkwave/gtkwave/blob/master/gtkwave3-gtk3/examples/gtkwaverc
https://github.com/gtkwave/gtkwave/blob/master/gtkwave4/examples/gtkwaverc

OpusElectronics, 2023-06-25:
@rfuest has reproduced the problem, but I don't know on which OS he built it (if that matters) nor with which configure options. (Default?) Did you use default configure options too?

Here's the Arch PKGBUILD file:
https://gitlab.archlinux.org/archlinux/packaging/packages/gtkwave/-/blob/main/PKGBUILD

If you're not familiar with PKGBUILD files, it should still be relatively self-explanatory. You can see which configure options are used, and they aren't the default.

gtkwave, 2023-06-26:
OK, I didn't try the gtk3 version which is why I didn't see this.
Try this.  I'll check in the fix into SF now.

-Tony

538:/tmp/gtkwave-gtk3-3.3.115/src> diff -c menu.c.1 menu.c
*** menu.c.1    2023-06-25 21:50:50.032667021 -0400
--- menu.c    2023-06-25 21:51:12.847768041 -0400
***************
*** 7415,7420 ****
--- 7415,7423 ----
      else
      {
      GLOBALS->show_base = gtk_check_menu_item_get_active(GTK_CHECK_MENU_ITEM(menu_wlist[WV_MENU_VSBS]));
+     if (!GLOBALS->signalarea || !GLOBALS->wavewindow) {
+                 return;
+             }
      GLOBALS->signalwindow_width_dirty=1;
      MaxSignalLength();
      signalarea_configure_event(GLOBALS->signalarea, NULL);

Message ID: ***@***.***