gtkwave/gtkwave #217: Feature request: configure or remember last window size

sschmitz86, 2023-07-03:
If you use GTKWave dozens of times during debug it can be really annoying to resize the window each time.
The default window is quite small.
It would be really nice to set a default in the settings menu or just remember the last size when GTKWave was closed.

gtkwave, 2023-07-03:
Save files remember window size.  Also see these rcvars for .gtkwaverc or specified at the command line:
       initial_window_x <value>
              Sets the size of the initial width of the wave viewer window. Values less than or equal to zero will set the initial width equal to -1 which will let GTK determine the minimum size.

       initial_window_xpos <value>
              Sets the size of the initial x coordinate of the wave viewer window. -1 will let the window manager determine the position.

       initial_window_y <value>
              Sets the size of the initial height of the wave viewer window. Values less than or equal to zero will set the initial width equal to -1 which will let GTK determine the minimum size.

       initial_window_ypos <value>
              Sets the size of the initial y coordinate of the wave viewer window. -1 will let the window manager determine the position.

-Tony

sschmitz86, 2023-07-05:
Thanks a lot, i did not knew that. (well, i should read the documentation to the end...)