gtkwave/gtkwave #218: Use native window decroations on X11 instead of ugly gnome decorations

plomari, 2023-07-19:
Since a while ago, gtkwave started drawing its own window decorations (title bar, window borders) on X11. It requests the window manager not to add any, probably using _MOTIF_WM_HINTS.

I suspect this happened with the change to GTK 3 and by the addition of a call to gtk_window_set_titlebar(). Removing gtk_window_set_titlebar() should stop this behavior.

It seems this behavior can't be disabled by the user in any way. It's extremely impractical because it breaks window manager features. It would be nice if this could be fixed.

rfuest, 2023-07-19:
Try to comment out this line and see if it has the desired effect: https://github.com/gtkwave/gtkwave/blob/master/gtkwave3-gtk3/src/gtk23compat.h#L20
Maybe you also need to set the `GTK_CSD` environment variable to `0`.

rfuest, 2023-07-20:
> I've built the `gtkwave3-gtk3` sub directory with `./configure --enable-gtk3` and can confirm that commenting this line brings back proper window decorations and restores the menu bar (which is much saner than the "burger" button).

While I hate burger menus on websites when I visit them on a desktop PC, I'm not against burger menus in application. But the menu must be designed to be used as a burger menu and should only contain a limited number of menu items. The deeply nested GTKWave menu works much better as a traditional menu bar.

gtkwave, 2023-07-20:
Note that I hate those menus as well.  That was more along the lines of "this is how an application should present X" from the gtk style guide that I saw somewhere on gtk.org.

-Tony

plomari, 2023-07-20:
> But the menu must be designed to be used as a burger menu and should only contain a limited number of menu items. The deeply nested GTKWave menu works much better as a traditional menu bar.

To be honest, whether you like GNOME's UI approach is probably a matter of taste. (Full disclosure: I dislike it very much and I don't use GNOME, so gtkwave looks weird on my desktop.)

But this is a good argument to make. gtkwave doesn't even make good use of this UI concept, so it's a disimprovement even on the GNOME desktop.

rfuest, 2023-07-20:
> To be honest, whether you like GNOME's UI approach is probably a matter of taste. (Full disclosure: I dislike it very much and I don't use GNOME, so gtkwave looks weird on my desktop.)

I like GNOME and have been using it for many years, but I don't necessarily agree with all their UI decisions. Some applications are too much designed like mobile apps IMO.

> But this is a good argument to make. gtkwave doesn't even make good use of this UI concept, so it's a disimprovement even on the GNOME desktop.

We all seem to agree that the burger menu, in it's current form, is a bad idea and I think we should get rid of it. I might revisit the idea for GTKWave 4, but not without massive changes to the menu. And I'll try to keep other DEs in mind when I design the new UI.

rfuest, 2023-07-20:
> That was more along the lines of "this is how an application should present X" from the gtk style guide that I saw somewhere on gtk.org.

Most of the GTK docs are biased towards the Gnome way of doing thing. And I'm not sure if there is or was a separate GTK style guide or only the Gnome HIG.

rfuest, 2023-08-16:
The master version doesn't use a headerbar any more and the menu button has been replaced by a regular menubar.

plomari, 2023-08-16:
Good news, thanks!