gtkwave/gtkwave #219: Creating releases from the GitHub version

rfuest, 2023-07-20:
At the moment the only available packaged versions of the GitHub version of GTKWave are the nightly windows builds. I think it would be good to also release the GitHub version to users. A GTKWave 4 release is still a long way out and IMO it's a good idea to fix some issues with GTKWave 3 in the meantime.

One example, that came up recently, is the Wayland support in GTKWave 3. I've recently switched to Wayland and would like to get GTKWave 3 working properly working on Wayland. But doing this correctly will require extensive code changes that won't be able to be included in the SourceForge version immediately. And I don't want to spend any time on this if no user will be able to benefit from this work.

My suggestion to solve this problem would be to release two versions of GTKWave. One regular version from the code that is on GitHub and a LTS release which is based on the SourceForge code. At the moment most users will not be aware that two versions exist and adding a LTS suffix should make the distinction between the versions clear.

My plan to implement this change:
- Move all development to GitHub
  The three subdirectories for the different GTKWave versions will be replaced by three branches:
  - `master`: Contains the content of the `gtkwave3-gtk3` directory. Any further development of GTKWave 3 will happen on this branch.
  - `lts`: Contains the SourceForge version of GTKWave. Only important fixes will be cherry picked from the `master` branch.
  - `gtkwave4`: Contains the development version of GTKWave 4.  
  - The `gtkwave3` directory will be deleted, because it's outdated anyway.
- The `master` branch will be released as "GTKWave 3.4.xxx".
- The stable version, which had been maintained on SourceForge, will be renamed to "GTKWave LTS" and will continue to be released as "3.3.xxx".
- GIT tags and GitHub releases will replace the tar files on SF.

umarcor, 2023-07-20:
IMHO, changing to the three branch structure would be desirable regardless of making the change to GitHub as the main location. I am also in favour of making GitHub the main location, but I wouldn't like that to be a stopper for the structure reorganisation.

With regard to generating packaged versions for other platforms and/or generating packages for multiple branches/versions, please let me know if you need help setting up CI.

gtkwave, 2023-07-20:
My suggestion to solve this problem would be to release two versions of GTKWave. One regular version from the code that is on GitHub and a LTS release which is based on the SourceForge code. At the moment most users will not be aware that two versions exist and adding a LTS suffix should make the distinction between the versions clear.
Message ID: ***@***.***>
That sounds fine to me.  I need to kick out the .116 point release yet (maybe this weekend?) and then I'll be ready to go into LTS maintenance mode.

3.3.116 25jun23 Fix manpage/odt for vcd2fst command switch documentation for zlibpack.
        Add GDK_WINDOWING_WAYLAND check for gdkwayland.h header usage.
        Changed sprintf to snprintf in fstapi.c.
        Fix init crash on show_base_symbols enabled.

They're trivial fixes, but it's time I got them out the door.

-Tony

rfuest, 2023-07-21:
I've created a LTS branch from the SVN history on SourceForge, which I'm planning to push to the main GTKWave repo later (see https://github.com/rfuest/gtkwave/tree/lts). The first SVN revision included in the LTS branch is 1545, which matches the initial GIT commit. It's not ideal that both branches don't have a common history and diverge from the initial commit, but at least they will be included in the same repo.

There are also tags for all versions from 3.3.104 to 3.3.115 (see https://github.com/rfuest/gtkwave/tags) which I have verified to be identical to the TAR files on SF. 3.3.112 - 3.3.114 are excluded, because the code doesn't seem to be in the SVN repo.

rfuest, 2023-07-21:
> With regard to generating packaged versions for other platforms and/or generating packages for multiple branches/versions, please let me know if you need help setting up CI.

Thanks for the offer. As a first step we need to fix the current CI setup after PR #220 is merged. Perhaps you could take a look at that after the PR is merged.

A nightly Linux build would also be nice to have. I've seen some projects that release nightly builds of the Flatpak version as their own Flatpak repo. This would have the advantage that one version would be usable on all distros, but I haven't looked into how complicated this would be to set up.

I don't have any idea what would be involved in building the macOS version in CI or the state of the macOS version in general. While I would like to take a look at fixing the macOS support I don't have access to any current Mac hardware with a current macOS version. And even a Hackintosh VM isn't really enough anymore since the switch to ARM.

When the `gtkwave4` branch is updated it will require a different CI setup, but I'll contact you once it is uploaded to discuss the necessary changes. Having easily accessible nightly builds of that branch will IMO be more important than the relatively stable `gtkwave3-gtk3` version to get users to test the new version.

rfuest, 2023-07-24:
The `gtkwave3-gtk3` directory is now the new root of the repository and the `lts` branch with tags for previous versions is also created.

rfuest, 2023-07-24:
@umarcor: Is there any documentation for users how they can use the nightly Windows builds? A regular user might be confused by the files on the nightly release page and unsure what they should pick. Some docs on the release page or the project README would make the Windows nightly builds easier to use.