gtkwave/gtkwave #227: "Open Source Definition" segfaults

rfuest, 2023-08-08:
If I try to use "Open Source Definition" on any signal in the FST file that was posted in this comment https://github.com/gtkwave/gtkwave/issues/173#issuecomment-1329965360 GTKWave segfaults. The problem is caused by an invalid `stem_path_number` in this part of the code: https://github.com/gtkwave/gtkwave/blob/8e0bcdb2e3df3078133e9bdf8d326b455f7c1ca5/src/fst.c#L400-L426 `stem_path_number` is zero and `stem_idx` overflows, because it is decremented by one. I'm not sure if the `stem_path_number` index is supposed to be 0 or 1 based and whether the error is in GTKWave or the FST file writer.

gtkwave, 2023-08-08:
IIRC, stems are supposed to start at one.Look in fstWriterSetSourceStem_2() in fstapi.c where sidx = ++xc->path_array_count.
I'd ask if this is an issue with _WAVE_HAVE_JUDY enabled/disabled in the writer when the FST file was generated in Icarus.  I don't know if that has been tested properly.
It's either that or something weird like JudyHSIns succeeded on some invocation because a null string or something similarly weird was passed to it by Icarus client code.

I know this code has worked before, so anything that broke is likely configuration related in a code path different from what I've used.

-Tony

gtkwave, 2023-08-08:
It looks to me like one of the buffer overflow hygiene fixes a user signaled worked great for the usual reading case except here the stems readout is wrong.  When I replace helpers/fst/fstapi.c with the version from gtkwave-3.3.100, it works fine.
This is one reason I hate touching code that works in the name of making it "better": it can get broken.  :-)

-Tony

gtkwave, 2023-08-08:
That's it.  A security fix I put in for 3.3.114 broke function as I didn't fully verify the fix in that function.

I'm glad it was that easy.  Thanks for finding and reporting that.

-Tony

538:/home/bybell/git/gtkwave/src/helpers/fst> git push origin master
Username for 'https://github.com': gtkwave
Password for ***@***.***': 
Counting objects: 11, done.
Delta compression using up to 4 threads.
Compressing objects: 100% (6/6), done.
Writing objects: 100% (6/6), 528 bytes | 0 bytes/s, done.
Total 6 (delta 5), reused 0 (delta 0)
remote: Resolving deltas: 100% (5/5), completed with 5 local objects.
remote: To https://github.com/gtkwave/gtkwave
   8e0bcdb..eda3e27  master -> master

768:/home/bybell/gtkwave/code/gtkwave3-gtk3/src/helpers/fst> diff -cNb fstapi.c.1 fstapi.c*** fstapi.c.1    2023-08-08 19:32:48.800412144 -0400
--- fstapi.c    2023-08-08 19:37:36.924954216 -0400
***************
*** 4181,4187 ****
                                  if((xc->hier.u.attr.subtype == FST_MT_SOURCESTEM)||(xc->hier.u.attr.subtype == FST_MT_SOURCEISTEM))
                                          {
                                          int sidx_skiplen_dummy = 0;
!                                         xc->hier.u.attr.arg_from_name = fstGetVarint64((unsigned char *)xc->str_scope_nam, &sidx_skiplen_dummy);
                                          }
                                  }
                          break;
--- 4181,4187 ----
                                  if((xc->hier.u.attr.subtype == FST_MT_SOURCESTEM)||(xc->hier.u.attr.subtype == FST_MT_SOURCEISTEM))
                                          {
                                          int sidx_skiplen_dummy = 0;
!                                         xc->hier.u.attr.arg_from_name = fstGetVarint64((unsigned char *)xc->str_scope_attr, &sidx_skiplen_dummy);
                                          }
                                  }
                          break;

Message ID: ***@***.***>

closed 2023-08-09 by rfuest