gtkwave/gtkwave #230: Extending the waveform display by one tick

rfuest, 2023-08-17:
At the moment the waveform display stops at the last transition in the dump file, if the end time isn't extended by another simulation time command (in case of a VCD file). This makes it hard to see the value of the last transition in a file and especially for vectors the value is only visible by placing the primary marker over it.

I would suggest to change the displayed time range to the half-open interval `[start_time, end_time+1)`. If the file contains all transitions up to `end_time` the values of the variables must be stable until at least `end_time+1` and it should be unambiguous to extend the display up to, but not including, `end_time+1`.

Before:
[image]

After:
[image]

Note that only the display of the waveform is extended by one tick. The markers are still limited to the the interval `[start_time, end_time]` and there is no timebar label for `end_time + 1`.

gtkwave, 2023-08-17:
It's still effectively going to be invisible if the timescale is set to ps and the sim clock is in ns, etc because nobody is ever going to zoom in to the ps level.

The way I've dealt with this is to click on the signal then arrow right to the very last signal transition.
-TonyMessage ID: ***@***.***>

rfuest, 2023-08-17:
Sure, it won't really help in that case. But would extending the range have any negative effects?

My use case is to view files which only contain a few cycles, like the output of a formal verification tool. In this case you are zoomed in far enough that each time tick is visible.

gtkwave, 2023-08-17:
Extending the range shouldn't have any negative effects.  Values maintain their value until changed and we're a long way away from the max time value.
-Tony
Message ID: ***@***.***>

cr1901, 2024-04-18:
>It's still effectively going to be invisible if the timescale is set to ps and the sim clock is in ns, etc because nobody is ever going to zoom in to the ps level.

I've been running into this issue with [Amaranth's](https://github.com/amaranth-lang/amaranth) VCD generation, which uses a timescale of femtoseconds. I tried [a local hack](https://libera.irclog.whitequark.org/amaranth-lang/2024-04-18#1713408137-1713409733;) to extend the VCDs manually by 1000 femtoseconds, but it's still too small of an extension to see the last transitions in the dump file until I zoom in to the ps/fs level. These are for simulations with a 12MHz clock! I think extending the range to the edge of the screen would be much more helpful.

tbybell, 2024-04-19:
You can experiment, but in button_motion_common()
```
    if (xin > (GLOBALS->wavewidth - 1))
        xin = (GLOBALS->wavewidth - 1);
```
...do you get near the behavior you want if the -1s are removed from this code of if those are replaced with +1/+2/etc?  Of course there may need to be additional handling such that this is only in effect in the vicinity of the end of simtime values, but does that make the transition you want reachable?

You will have to comment this out as well:
```
if(newcurr>GLOBALS->tims.last)          /* sanity checking */
        {
        newcurr=GLOBALS->tims.last;
        }
```