gtkwave/gtkwave #231: No feedback to the user when loading fails

rfuest, 2023-08-18:
If GTKWave is started from a graphical file manager there is no feedback to the user if loading fails, for example, when the dumpfile that belongs to a `gtkw` file is missing. In this case GTKWave should show a message box instead of relying on console output.

The same feedback is also missing when a file is opened via "File -> Open New Window" or "File -> Open New Tab" or via drag and drop. Trying to open a broken file in a new tab does also close the entire window with all tabs.

gtkwave, 2023-08-18:
The fun part of this is in the 1.x version, there used to be a status pane in the upper left that communicated such info but it was eventually removed. 

I do agree though, if there is no visible attached terminal, such feedback needs to go somewhere.
-Tony
Message ID: ***@***.***>

rfuest, 2023-08-19:
> The fun part of this is in the 1.x version, there used to be a status pane in the upper left that communicated such info but it was eventually removed.

I do actually remember using a version that had the messages and the old style buttons, but it has been a while. Some form of showing the console output inside the UI would still be useful. The gerbv Gerber file viewer from the gEDA project does, for example, show the messages in another tab in the sidebar and automatically changes to that tab when import errors occur.

> I do agree though, if there is no visible attached terminal, such feedback needs to go somewhere.

As a first step a simple message box should be enough. In the future this could be extended to allow the user to pick another dump file, if the error was caused by opening a `gtkw` file where the associated dump file couldn't be found.