gtkwave/gtkwave #236: Non-aesthetic rendering

dov, 2023-08-22:
The rendering of analog signals is less than optimal. Compare the old.img which is the current gtkwave rendering vs new.img which is my experimental branch. 

There are mainly two issues:

1. The rounding of drawing coordinates to integers before they are passed to cairo. 
2. The use of the pattern MLSMLS... (where M=MoveTo, L=LineTo, and S=Stroke) which causes thin lines to be to thick.

I hope to send a PR in the next few days:

Current rendering:
[image]

My experimental rendering:
[image]

rfuest, 2023-08-22:
That looks much better, I hadn't realized that you were talking about analog waveforms. For analog it's also easier to implement the `MLLLLLLL...S` pattern, because it is one continuous line without changes in color, except for special values like NAN and INF.

> I hope to send a PR in the next few days:

I would suggest that you wait until the new drawing code is merged, it might even have a similar fix already implemented, but I haven't looked at that code in over a year. At least the conversion from time to coordinates is using floating point.

dov, 2023-08-22:
Ok, I won't create a PR for now. But for the record, the branch that did the above rendering is here:

https://github.com/dov/gtkwave/tree/DrawWithDouble