gtkwave/gtkwave #239: Example file repository

rfuest, 2023-08-23:
When I'm testing GTKWave code changes I often need to find an appropriate test file for specific features. I think it would be a good idea to have a separate repository to collect files that can be used to test GTKWave. This collection would not only be handy for development, but also when diagnosing bugs. Not every user can share the dump files they are working on when they report a bug and having a commonly available test file can help with the debugging process.

I wouldn't want to include these files in the main repository, because there should at least be some larger files in the example file repository, which would unnecessary bloat the source code repo.

Some ideas of what could be included:
- A basic dump file with a few signals in all supported dump file formats
- A save file which contains all trace types (like groups, blanks, analog, colored traces, comments, ...)
- Demos for all filter types
- A FST file with stems and the matching source file
- Different larger design to diagnose performance problems, for example:
  - One design with not too many signals, but a lot of transitions
    The file from #173 was already useful to improve the performance of the drawing code. A file with similar characteristic, but perhaps a bit longer would be a good addition.
  - One design which is heavily nested
    A while ago a user posted a design which consisted of a lot of modules (standard cells) and it showed performance issues in the parsing or tree generation code.
- VCD files from different tools
  e.g. for testing the special case that was included for Icarus VCD files
- Examples how people use the TCL interface
- Anything else which demonstrates/test features of GTKWave I didn't think about

Any user contributions to this repo would be very appreciated. All contributed files will need to be under some open source license.

umarcor, 2023-08-28:
I was just thinking that it would be good to have performance tests. Since the codebase is being updated/reworked, those tests could help know that peformance is maintained (or improved) while the codebase is refactored.

In https://github.com/umarcor/osvb/tree/main/sigrok, there is a script to generate waves with GHDL (see https://github.com/umarcor/osvb/blob/main/sigrok/resolution/run.sh), which would make pulseview/sigrok choke. See https://umarcor.github.io/osvb/notebook/sigrok.html#performance.

With regard to creating a secondary repo, I believe Tony would need to do it and then make you a contributor.
Alternatively, a gtkwave organisation could be created, to contain both repos.
I believe there are two possible procedures:
- Change his username to say 'tonybybell', immediately create org 'gtkwave' and transfer this repo from his user namespace to the org.
- Create user 'tonybybell' (probably with a different e-mail account), then convert 'gtkwave' to an org (making tonybybell the admin).

Both solutions are explained in https://docs.github.com/en/account-and-profile/setting-up-and-managing-your-personal-account-on-github/managing-your-personal-account/converting-a-user-into-an-organization.

rfuest, 2023-08-28:
> I was just thinking that it would be good to have performance tests. Since the codebase is being updated/reworked, those tests could help know that peformance is maintained (or improved) while the [...]

Thanks for the links. The main purpose for the repo I suggested in this issue is manually testing of performance issues and features. At a later date, when more of `libgtkwave` gets merged, there will also be automated test, for which I've already created many synthetic test files. I could add some of these test and the framework to create them to the new repo.

For the performance tests I would prefer to have some dump files for real designs. I think it is better to analyse the drawing performance based on more realistic designs and not only rely on synthetic tests. If a particular area of GTKWave is identified to have performance issue these designs could be augmented with additional synthetic tests to focus on these issue.

Would you be interested in contributing some GHDL based designs to generate some examples? VHDL based designs and GHW file support has not been tested as well during my refactoring. Something similar to the script you linked and a simple design that contains a few different scope and data types would already be helpful. And maybe the output of simulating a larger open source project to get a dump file for performance testing.

> With regard to creating a secondary repo, I believe Tony would need to do it and then make you a contributor. Alternatively, a gtkwave organisation could be created, to contain both repos. I believe [...]

I've discussed this with Tony before and he agreed to change `gtkwave` into a organization, but he didn't have time to change it yet.

gtkwave, 2023-08-28:
Thanks for the link on converting to an org.  I'll look into it later this week or this upcoming weekend if I can.  I am extremely busy this week, which unfortunately seems to be a current theme lately...

-Tony
Message ID: ***@***.***>

rfuest, 2023-08-28:
> Thanks for the link on converting to an org.  I'll look into it later this week or this upcoming weekend if I can.  I am extremely busy this week, which unfortunately seems to be a current theme lately...

If you find some time to get this done it would be great, but there is no rush. I'll be away in the next two weeks and won't be able to create any new repositories until the second half of September.