gtkwave/gtkwave #241: Bug in interpolated analog signal, or request for linear connection

dov, 2023-08-24:
Consider the desire to draw a pair of signals corresponding to the velocity and the position of the following system:

[image]

The system contains 6 time points and I express them in a vcd (analog) file as follows:

```
$date   2023-08-24_17-05-30 $end
$version    0.1.0 $end
$comment   A bug? $end
$timescale 1s $end
$scope module Motion $end
$var real 64 x x $end
$var real 64 v v $end
$upscope $end
$enddefinitions $end
$dumpvars
r0 x
r0 v
#0 r0  x r0  v
#1 r0  x r-1 v
#2 r-1 x r0  v
#3 r-1 x r1  v
#4 r0  x r0  v
#5 r0  x r0  v
```

I load them into gtkw and choose analog step display for "v" (since its change is instanteneous) and analog interpolation for "x". The result is the following:

[image]

What I would have liked to see is x as a simple line between the coordinates:

```
0 0
1 0
2 -1
3 -1
4 0
5 0
```

Is this a bug in the "interpolation" (how many points does it use), or would it make sense to add an analog "line" display, that simply does a lineto between points without any additional processing?

rfuest, 2023-08-24:
This is caused by the VCD import, which automatically removes transitions to identical values from the input. If you enable the "vcd_preserve_glitches_real" rc variable you'll get the desired result.

dov, 2023-08-24:
Thanks. That is certainly not something that I would have guessed. I wonder why it is needed.

I'll probably turn it on by default in the installer for my coworkers.

rfuest, 2023-08-24:
I would like to get @gtkwave's input on this. Turning the option to `on` would be a better default IMO, because this behaviour is confusing. But in the end this would only be a partial fix, because the same issue exists if you draw a vector as an analog waveform.

gtkwave, 2023-08-24:
Gtkwave is a digital wave viewer.  Analog with respect to a continuously varying signal is a bolted-on afterthought and the implementation is not something like SPICE would show.

Analog drawing was added here in 2005:
Analog support is Copyright (C) 2005-2023 Thomas Sailer.

I believe Thomas was doing some kind of PLL or similar simulation and what gtkwave renders based upon his patches (the hptr/vptr analog functions) was good enough for his needs.  I myself use analog with step (not interpolation) simply to measure varying population counts on valids for things like queue entries and cache tags, so I never see interpolation misrepresentations.

I don't really care what that glitches real removal value is set to.  It's probably enabled simply to cut down on memory usage for poorly written dumpers that write out values every timestep whether a signal changes value or not.

An interpolation reader that performs so-called deglitching really should only start removing duplicate values such that only the endpoints in a train of 3+ consecutive values are preserved.  Now we're in trace loader territory across all sim dump formats having to change that 2nd value's time stamp to the 3rd one.  That said, viewing an integer vector as a double/float analog wave will not behave as you expect as intermediate values will be removed.  Those values are irrelevant in the digital domain, but not when you're interpolating to produce a curve.  My guess is that rc variable was created when a user observed the same problem on doubles long ago.

-TonyMessage ID: ***@***.***>

dov, 2023-08-25:
In light of @gtkwave's answer I will create a pull request that changes the default of `vcd_preserve_glitches_real` and `vcd_preserve_glitches`  to true.