gtkwave/gtkwave #246: Multiple buffer overflows in LXT/LXT2/VZT parsing

ArcaneNibble, 2023-09-03:
While trying to write [my own documentation for these file formats](https://github.com/ArcaneNibble/lxt/blob/main/format_notes.md), I have discovered multiple buffer overflows and similar invalid memory access issues in GTKWave file parsers:

# LXT

- [ ] If `total_mem` in the `LT_SECTION_FACNAME` section is too small, a heap buffer overflow occurs
- [ ] If a `LT_SYM_F_ALIAS` facility contains an out-of-bounds index, invalid memory is accessed
- [ ] If a MVL_2 change command references a dictionary entry, and that dictionary entry is longer than it is supposed to be (i.e. longer than the facility length), memory is corrupted with '0' characters until a segfault occurs

# LXT2

- [ ] If `longestname` in the header is too small, a heap buffer overflow occurs
- [ ] If `num_time_table_entries` in a granule is larger than 64, the fixed-size `time_table` array in the `lxt2_rd_trace` struct is overflowed

# VZT

- [ ] If `longestname` in the header is too small, a heap buffer overflow occurs

gtkwave, 2023-09-04:
I'm sure there are all sorts of buffer overflows.

Writer client code is supposed to know what it is doing.  :-)  Joking aside, these aren't a general purpose libc functions so I'm not concerned about security implications, but crashes are no good.  I would be surprised if more than a dozen people have ever interfaced to those writers unless a class project was involved.
If you want, open a bug so they're documented and can be fixed when there's time.  With well-behaved writers (such as simulators that write those formats, e.g., Icarus), I don't expect any crashes from them.

Thanks,
-Tony
Message ID: ***@***.***>

ArcaneNibble, 2023-09-09:
The reason I've opened this issue is because every single one of the bugs here can cause GTKWave to crash (even though I agree that the security implications are not too important for most people for normal use cases).

I've attached a ZIP file containing a test case for each of these crashes, all of which reliably crash GTKWave 3.3.107 (at least on macOS aarch64, installed via Homebrew)

[crashes.zip](https://github.com/gtkwave/gtkwave/files/12566000/crashes.zip)

I have also discovered a few new ways to cause crashes which I have edited the issue description to include. Incidentally, I _accidentally_ discovered the "If `srclen` exceeds the largest `dstlen`" bug while trying to make a valid, non-corrupted, non-malicious file, so fixing all of these bugs would be a nice robustness improvement.

gtkwave, 2023-09-09:
Thanks.  

Do existing simulators generate corrupt files that exhibit this behavior?  LXT/LXT2/VZT are quite old formats (2001/2003/2003) and in my opinion aren't worth fixing and should be removed at some point.  They have some historical interest in various ways of storing sim data as nobody else* really has documented any of this anywhere, but they're been surpassed in the general case for size and performance by FST.

(*) On my travels, I've encountered a handful of interesting papers by Edwin Naroska, but that was it.  LXT/LXT2/VZT were inspired by the data representations found in internal IBM formats from the late 1990s.

-Tony
Message ID: ***@***.***>

ArcaneNibble, 2023-09-09:
Nope, no "real" tool that I am aware of generates corrupt files like this. I only stumbled upon these bugs while trying to write my own tooling for them (and to understand what specific optimizations these formats use).

I wasn't aware that there are vague plans to deprecate these formats. Perhaps this could somehow be mentioned more clearly in the manual?

rfuest, 2023-09-29:
> I wasn't aware that there are vague plans to deprecate these formats. Perhaps this could somehow be mentioned more clearly in the manual?

I did just remove support for the formats from the development version of GTKWave to reduce the amount of code that needs to be refactored in the future. The documentation will still need to be updated to reflect this change and I've opened a issue to track the necessary changes #256.