gtkwave/gtkwave #248: xml2stems show lines 0

needhnlxf, 2023-09-08:
I use the xml2stems to get a stems file. But I find the lines is 0 in file.

Such as below:
++ module tb_top_verilator file  lines 0 - 0
++ begin load_prog file  line 0
++ comp load_prog:G0 type load_prog:G0 parent tb_top_verilator
++ module load_prog:G0 file  lines 0 - 0
++ endbegin load_prog
++ begin unnamedblk1 file  line 0
++ comp unnamedblk1:G1 type unnamedblk1:G1 parent tb_top_verilator
++ module unnamedblk1:G1 file  lines 0 - 0
...................................................

Then it can't open the rtl source file in rtlbrowser. It shows "Could not open sourcefile 'lines'".
(gtkwave version is 3.3.117)

Can you help me? Thanks!

gtkwave, 2023-09-08:
Did verilator change something?  There should be a filename listed in there.  You'll have to check with Wilson Snyder on this.

-Tony
Message ID: ***@***.***>

needhnlxf, 2023-09-11:
I don't know what happen. I use the verilator version is 5.008. And the xml file is below.
[Vtb_top_verilator.zip](https://github.com/gtkwave/gtkwave/files/12571818/Vtb_top_verilator.zip)
Is it right?

gtkwave, 2023-09-12:
| I don't know what happen. I use the verilator version is 5.008. And the xml file is below.
| Vtb_top_verilator.zip
| Is it right?
See if the attached fixes your problem.  It looks like the file tag changed from "fl" to a different "loc" format in later versions of Verilator.

-Tony

  Message ID: ***@***.***>

needhnlxf, 2023-09-12:
So, you mean just change the loc to fl in xml file will may generate correct stems file?

gtkwave, 2023-09-12:
No.  I uploaded a replacement xml2stems.cc file in the github message thread.  Please recompile against that.
-Tony

needhnlxf, 2023-09-13:
Well, how can I get it? Is there a link?

gtkwave, 2023-09-13:
https://github.com/gtkwave/gtkwave/files/12581744/xml2stems.zipMessage ID: ***@***.***>
...github's email feature much to my consternation isn't complete and they attempt to steer you to using their web interface.
Copy this over the existing one in your gtkwave source distribution, then rebuild.

-Tony

needhnlxf, 2023-09-14:
There is something wrong with the link. When I click it, it says page not found. 
Can you send again? Thanks!

gtkwave, 2023-09-14:
click on "view it on GitHub" in the email message
look two days ago for the zip file link found in one of my responses.

-Tony
Message ID: ***@***.***>

needhnlxf, 2023-09-14:
Ok, it is work fine. 
Thanks!

closed 2023-09-14 by needhnlxf

gtkwave, 2023-09-14:
Fixed in:
remote: To https://github.com/gtkwave/gtkwave
   994b1ab..0474cff  master -> master