gtkwave/gtkwave #250: Unable to launch GTKWave on macOS Sonoma

andylin2004, 2023-09-15:
[image]

This prompt will show up, and then the app will close out

gtkwave, 2023-09-15:
Simply showing a screenshot doesn't convey much information.  What is the version of gtkwave and where did you get it?
Is this with respect to the binaries grabbed off the sourceforge site?  If so, sorry, I don't own a Mac.  Those binaries were compiled on a machine running a Snow Leopard (10.6) VM created from back around 2011 or so.  I stopped creating binaries when it looked like compatibility issues would start occurring.

Thanks,
-Tony
Message ID: ***@***.***>

andylin2004, 2023-09-15:
I am using version 3.3.107, installed via brew on macOS.

gtkwave, 2023-09-17:
OK, you're using the most recent version that I built...which was three years ago.
As I said, I don't have a Mac, so unfortunately, I can't build a binary that will work for you.  Setting up a build environment on Mac using GTK-OSX is a fairly involved process unless it's been simplified.  Another user might be able to help you.

-Tony
Message ID: ***@***.***>

randomplum, 2023-09-27:
@andylin2004 

if you're using brew you can try:

class Gtkwave < Formula
  desc "GTKWave"
  homepage "https://gtkwave.sourceforge.net"
  license "GPL-2.0-or-later"
  head "https://github.com/randomplum/gtkwave.git", branch: "macos_sonoma"

  depends_on "desktop-file-utils" => :build # for update-desktop-database
  depends_on "shared-mime-info" => :build 
  depends_on "gobject-introspection" => :build
  depends_on "meson" => :build
  depends_on "ninja" => :build
  depends_on "pkg-config" => :build
  depends_on "gtk+3"

  def install
    ENV["DESTDIR"] = "/"
    system "meson", "setup", "build", *std_meson_args
    system "meson", "compile", "-C", "build", "--verbose"
    system "meson", "install", "-C", "build"
  end

end

save it in gtkwave.rb file and run brew install --build-from-source --HEAD ./gtkwave.rb

This is the only needed part of the patch I have in the fork
[image]

There is however a problem, on Apple silicon based laptops GTKWave only displays waveforms correctly on the external screen. when using build in screen, only 1/4 of the waveform window renders waveforms

[image]

I'm looking into it, but any ideas on what's happening are welcome.

rfuest, 2023-09-28:
I'm kind of surprised that GTKWave build on macOS at the moment, because I didn't have any way of testing it during the rewrite. But it's nice to see that it still works. The same regex line did also cause issues in some Windows builds in the past.

> There is however a problem, on Apple silicon based laptops GTKWave only displays waveforms correctly on the external screen. when using the build in screen, only 1/4 of the waveform window renders waveforms
>
> I'm looking into it, but any ideas on what's happening are welcome.

This probably has something to do with display scaling, maybe the image surface isn't large enough. You can try to manually increase the size here, to try to debug the issue: https://github.com/gtkwave/gtkwave/blob/a2cb676327193e37bf99615d0388d752938b170c/src/wave_view.c#L701C1-L706C86
But the entire drawing code will be replaced soon and I would suggest to not spend too much time on trying to fixing this.

dpretet, 2023-09-28:
Hello,

I have the same problem than @andylin2004 and would be glad to keep GTKWave in my workflow If possible. Thanks @randomplum for the fix :) It works also on my side with the same 1/4 issue. I use a mbp m1 2020 if useful for anybody.

Damien

randomplum, 2023-09-28:
@rfuestThat was it
https://github.com/gtkwave/gtkwave/blob/0fe81d1ff3f0cb688f520cf8003c854c51124f31/src/wave_view.c#L700
I'm not familiar enough with gtk to know of it's a proper way to do it, but https://docs.gtk.org/gdk3/method.Window.create_similar_image_surface.html points to scaling width and height you feed to gdk_window_create_similar_image_surface.

I was also surprised that it worked so easily (I even got it to build a proper .app without homebrew, but this takes quite a bit more hacking to get for not much gain), good job!  

Would You accept this and regex_t with the __APPLE__ if def ? 
Unless this code will be gone by the time 3.4.0 get's released.
Either way I'll try to push the homebrew formula when the 3.4.x get's released, if you're ok with it.

For anyone wanting to run on macos, I've updated my fork. if there is enough people interested I can setup a homebrew tap.

andylin2004, 2023-09-28:
@rfuest @randomplum Thanks y'all for the help on getting this whole issue fixed! Sorry I couldn't be of help (since I'm relatively unfamiliar with the frameworks used in this project and it's academic season for me)

rfuest, 2023-09-28:
> I'm not familiar enough with gtk to know of it's a proper way to do it, but https://docs.gtk.org/gdk3/method.Window.create_similar_image_surface.html points to scaling width and height you feed to gdk_window_create_similar_image_surface.

Thanks for verifying this. I can't replicate the problem at the moment, because I don't have a HiDPI display and trying to enable scaling in Linux did just result in a blurry display.

One thing I noticed is that the current code actually modifies the passed `allocation` object, which might not be a good idea. You could try this code to see if it changes anything:
```c
    scale = gtk_widget_get_scale_factor(widget);

    g_clear_pointer(&self->traces_surface, cairo_surface_destroy);

    self->traces_surface =
        gdk_window_create_similar_image_surface(gtk_widget_get_window(widget),
                                                CAIRO_FORMAT_ARGB32,
                                                allocation->width * scale,
                                                allocation->height * scale,
                                                scale);
```

andylin2004, 2023-09-28:
Also for curiosity sake, how did you make a .app binary for gtkwave?

randomplum, 2023-09-28:
@rfuest Yes, I agree. Tested and it works correctly.
I've also setup a hombrew tap. Now it should be possible for everyone to just run
`brew install --HEAD randomplum/gtkwave/gtkwave`
and it'll build master branch gtkwave.

@andylin2004 macos .app are just directories with specific structure and a Info.plist file. The real fun starts when the app is using some libraries and you want to make it portable. You basically need to rebuild all the prerequisites against MacOS SDK and make library paths relative. Then for distribution you need to sign all the files, ideally with a nice apple developer program signature (99$/year) so you don't anger the Gatekeeper.

diogratia, 2023-09-29:
First off, I really appreciate what you guys have been doing. I have a slight problem with the result which may be from my naivete on the subject:

[image]

The size of the waveforms and text in the waves frame are scaled a bit small as is the text in the associated signals frame.

 I've prefixed invoking your tap

 `arch -x86_64 brew install --HEAD randomplum/gtkwave/gtkwave`

because I have both an arm64 Homebrew installation (for a native ghdl-llvm) and  an x86_64 installation in /usr/local for ghdl nightly builds both supplying supplying a llvm@15.

This is on a 2020 Macbook Air M1 running macOS 14.0.

`System Settings -> Displays -> Built-in Display -> 1680 x 1050` (which scales text to be readable on the 13.3 inch panel for compliant applications)`

Any hints on how to deal with the vertical height in the unlabelled waves and signals frames? I'd imagine that has to do with the ratio of the  internal'virtual'  display size and the actual screen size (2560 x 1600) of 0.64453125 on X and 0.65625 on Y.

Nothing showing up on stdout/stderr when running gtkwave or exiting.

randomplum, 2023-09-29:
@diogratia I think the problem can be that there is no Monospace font in macos, and that's the default (I have custom font in .gtkwaverc so I didn't notice). You can change font and size in the .gtkwaverc file (you most likely need to create one in user home directory).
I run this:
fontname_signals "Hack Nerd Font Mono" 16
fontname_waves "Hack Nerd Font Mono" 12

I use Hack Nerd Font which is not part of macos, but you can replace "Hack Nerd Font Mono" with "Menlo" which is macos monospace font.

randomplum, 2023-09-29:
@diogratia I found the problem. I've updated my tap to now build gtkwave with full mac integration support, this should make the default font correct. 
since it's head only formula tap to update you need to run:
`brew uninstall gtkwave`
`brew untap randomplum/gtkwave`
`brew install --HEAD randomplum/gtkwave/gtkwave`

@rfuest I've added some more changes to get the gtk-mac-integrations to work https://github.com/randomplum/homebrew-gtkwave/blob/11cd7d28cf0324f07dc9ec189b404c438fdaf7e8/macos_compat.diff

rfuest, 2023-09-29:
> @rfuest I've added some more changes to get the gtk-mac-integrations to work https://github.com/randomplum/homebrew-gtkwave/blob/11cd7d28cf0324f07dc9ec189b404c438fdaf7e8/macos_compat.diff

I've just applied the type change of  `preg_regex_c_1` to `regex_t` unconditionally. The same issue existed on Windows and I think the original code was incorrect. There should be no need for an ifdef anymore. I'll take a closer look at the rest of the changes when you open a PR for them.

> @rfuest @gtkwave let me know if you're open to a PR for improved macos support. I'm open to keep supporting it for as long as I'm using a mac.

Sure, any help is appreciated. Tony and I both have the problem that we can't support the macOS version, because we don't have access to the appropriate hardware. It would be great if you could take over support for the macOS version. At some point we should also setup CI to build a macOS version to at least get a notification if we did accidentally break macOS support in the future.

andylin2004, 2023-09-29:
> @diogratia I found the problem. I've updated my tap to now build gtkwave with full mac integration support, this should make the default font correct. since it's head only formula tap to update you need to run: `brew uninstall gtkwave` `brew untap randomplum/gtkwave` `brew install --HEAD randomplum/gtkwave/gtkwave`
> 
> @rfuest I've added some more changes to get the gtk-mac-integrations to work https://github.com/randomplum/homebrew-gtkwave/blob/11cd7d28cf0324f07dc9ec189b404c438fdaf7e8/macos_compat.diff
> 
> @rfuest @gtkwave let me know if you're open to a PR for improved macos support. I'm open to keep supporting it for as long as I'm using a mac.

I'm attempting to reinstall gtkwave via these instructions, but it doesn't seem to be working, with the terminal output showing this:
```sh
Treating gtkwave as a formula. For the cask, use homebrew/cask/gtkwave
==> Fetching randomplum/gtkwave/gtkwave
==> Downloading https://raw.githubusercontent.com/randomplum/homebrew-gtkwave/ma
######################################################################### 100.0%
==> Cloning https://github.com/gtkwave/gtkwave.git
Updating /Users/andylin/Library/Caches/Homebrew/gtkwave--git
==> Checking out branch master
Switched to branch 'master'
Your branch is up to date with 'origin/master'.
HEAD is now at d011156 Add GwColorTheme to manage waveform colors
==> Installing randomplum/gtkwave/gtkwave 
==> Patching
==> Applying macos_compat.diff
patching file meson.build
patching file 'src/cocoa/cocoa_misc.c'
patching file 'src/cocoa/cocoa_misc.h'
patching file 'src/main.c'
patching file 'src/meson.build'
No file to patch.  Skipping...
2 out of 2 hunks ignored--saving rejects to 'src/wave_view.c.rej'

If reporting this issue please do so at (not Homebrew/brew or Homebrew/homebrew-core):
  https://github.com/randomplum/homebrew-gtkwave/issues

gtkwave's formula was built from an unstable upstream --HEAD.
This build failure is expected behaviour.
Do not create issues about this on Homebrew's GitHub repositories.
Any opened issues will be immediately closed without response.
Do not ask for help from Homebrew or its maintainers on social media.
You may ask for help in Homebrew's discussions but are unlikely to receive a response.
Try to figure out the problem yourself and submit a fix as a pull request.
We will review it but may or may not accept it.
```

I'll stick to the manual install for now

krishnachaitanya1132, 2023-09-30:
> @diogratia I found the problem. I've updated my tap to now build gtkwave with full mac integration support, this should make the default font correct. since it's head only formula tap to update you [...]

Hello @randomplum 
I tried the brew install command as you mentioned in my apple M2 silicon with macOs 14 and facing error as below. can you let me if there is any workaround for this
[image]

randomplum, 2023-09-30:
@andylin2004  @krishnachaitanya1132 This is the problem when applying patches to an active repo.
I've pinned the tap to a forked repo so it's stable for now, until gtkwave gets to the release.

This is an experimental tap that builds git master branch, issues are expected.
please report tap issues here: https://github.com/randomplum/homebrew-gtkwave/issues

@andylin2004 I'm aware of the GTK asserts, I'll address them at some point in the future. It doesn't affect most of the application.

krishnachaitanya1132, 2023-09-30:
Can you let me know how to brew install using forked repo.
In the above run i just gave the brew install -HEAD cmd as you suggested.

Should I download the forked repo and install it as usual??

randomplum, 2023-09-30:
> Can you let me know how to brew install using forked repo. In the above run i just gave the brew install -HEAD cmd as you suggested.
> 
> Should I download the forked repo and install it as usual?? Using
> 
> * ./configure
> * make
> * sudo make install

@krishnachaitanya1132  The forked repo should be transparent.
All you need is to run those commands (if you didn't get it to install before just the last command should be enough)
`brew uninstall gtkwave`
`brew untap randomplum/gtkwave`
`brew install --HEAD randomplum/gtkwave/gtkwave`

diogratia, 2023-10-01:
@randomplum 

After finding the issue reported here I found your closed homebrew-gtkwave issue  [#1](https://github.com/randomplum/homebrew-gtkwave/issues/1) and figured from the description it was the same cause so didn't stir the pot further.

 I had noticed the File -> Grab to File has a scaling problem showing about a quarter of the gtkwave window and after your tap update allowing gtkwave to be built again the portion grabbed got smaller again while being enlarged. Not trying to push priorities, macOS users can screenshot the window just fine. 

I also noticed the font and size in the SST frame is different from the waves frame. Checking the gtkwaverc.5 man page and checking every (albeit a bit dated) .gtkwaverc file I have none of them contain fontname_logfile, fontname_signals or fontname_waves. I can easily imagine the defaults are different and now as shown.

Thanks for all the great work.

Khizarkk7, 2023-11-03:
GTK wave is not working macbook software: sonoma 14.1 
[image]
[image]
this are the pop kindly resolve this we are facing this problem macbook software: Sonoma 14.1

Khizarkk7, 2023-11-03:
> Simply showing a screenshot doesn't convey much information.  What is the version of gtkwave and where did you get it? Is this with respect to the binaries grabbed off the sourceforge site?  If so, [...]

**### From here**

[image]

andylin2004, 2023-11-03:
Have you attempted to install gtkwave using the brew formula that should be above?

Khizarkk7, 2023-11-03:
Brew formula is not working results are same same issue occurred update
gdkwaves version

randomplum, 2023-11-03:
Homebrew gtkwave does not replace the app, you need to run gtkwave binary that was build during the install, you can try it first with terminal, after that you can make a dock button for easy access.

Khizarkk7, 2023-11-03:
how can i install gtkwave binary kindly tell me the procedure in macbook
air m1 software: sonoma 14.1

terriblefire, 2023-11-07:
GTKWave needs rebuilt for MacOS Sonoma. Is there no CI setup for this? Since the Sonoma upgrade i'm completely unable to use GTKWave.

randomplum, 2023-11-07:
@Khizarkk7 @terriblefire  There is no working release for now. you can install it using homebrew and my tap with
```brew install --HEAD randomplum/gtkwave/gtkwave```

rfuest, 2023-11-07:
> GTKWave needs rebuilt for MacOS Sonoma. Is there no CI setup for this? ...

At the moment CI only checks that GTKWave can be built under macOS, but no nightly builds are generated. This is all I could setup without having access to a macOS machine. If someone wants to contribute the necessary CI changes we can start to add macOS binaries as build artifacts to CI runs.

ben-h-a, 2023-11-19:
@randomplum I'm getting the following error using your tap: 

homebrew version
`Homebrew 4.1.20`

MacOs version:
`
ProductName:		macOS
ProductVersion:		14.1
BuildVersion:		23B74
`

Output from cask:
`
brew install --HEAD randomplum/gtkwave/gtkwave -v
==> Fetching dependencies for randomplum/gtkwave/gtkwave: gobject-introspection and meson
==> Fetching gobject-introspection
==> Downloading https://ghcr.io/v2/homebrew/core/gobject-introspection/manifests/1.78.1-1
Already downloaded: /Users/benallen/Library/Caches/Homebrew/downloads/f6daa2936809f8f0f524f563231e536ec1acb5caaead8b2fd5a52ddf718a7f2c--gobject-introspection-1.78.1-1.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/gobject-introspection/blobs/sha256:025018b9b268379c594d7a079ea585962a82610632f5fa7682347d7cdb031755
Already downloaded: /Users/benallen/Library/Caches/Homebrew/downloads/b2eb72577fd14d94864c18d5d3d9f5d1f0103f326b17cadb96c6d5db566fa3b8--gobject-introspection--1.78.1.arm64_sonoma.bottle.1.tar.gz
==> Verifying checksum for 'b2eb72577fd14d94864c18d5d3d9f5d1f0103f326b17cadb96c6d5db566fa3b8--gobject-introspection--1.78.1.arm64_sonoma.bottle.1.tar.gz'
==> Fetching meson
==> Downloading https://ghcr.io/v2/homebrew/core/meson/manifests/1.2.3
Already downloaded: /Users/benallen/Library/Caches/Homebrew/downloads/ab5881810d8bcc8cbe3c3efcb7a180225aebb57333919ebabac22cd0888dcd76--meson-1.2.3.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/meson/blobs/sha256:f2609629267212ee0479482c3fb7d7c4a26b53e4a7059a18e08f494904ba93fa
Already downloaded: /Users/benallen/Library/Caches/Homebrew/downloads/b10f6e5f348dba175573f1064225421cb59fddcc4d350f1b204a1e02256cb3e2--meson--1.2.3.all.bottle.tar.gz
==> Verifying checksum for 'b10f6e5f348dba175573f1064225421cb59fddcc4d350f1b204a1e02256cb3e2--meson--1.2.3.all.bottle.tar.gz'
==> Fetching randomplum/gtkwave/gtkwave
==> Cloning https://github.com/randomplum/gtkwave.git
Updating /Users/benallen/Library/Caches/Homebrew/gtkwave--git
/usr/bin/env git config remote.origin.url https://github.com/randomplum/gtkwave.git
/usr/bin/env git config remote.origin.fetch \+refs/heads/master:refs/remotes/origin/master
/usr/bin/env git config remote.origin.tagOpt --no-tags
/usr/bin/env git config advice.detachedHead false
/usr/bin/env git config core.fsmonitor false
/usr/bin/env git fetch origin
==> Checking out branch master
/usr/bin/env git checkout -f master --
Already on 'master'
Your branch is up to date with 'origin/master'.
/usr/bin/env git reset --hard origin/master --
HEAD is now at d793aa1 Update io.github.gtkwave.GTKWave.metainfo.xml
==> Installing gtkwave from randomplum/gtkwave
==> Installing dependencies for randomplum/gtkwave/gtkwave: gobject-introspection and meson
==> Installing randomplum/gtkwave/gtkwave dependency: gobject-introspection
==> Downloading https://ghcr.io/v2/homebrew/core/gobject-introspection/manifests/1.78.1-1
Already downloaded: /Users/benallen/Library/Caches/Homebrew/downloads/f6daa2936809f8f0f524f563231e536ec1acb5caaead8b2fd5a52ddf718a7f2c--gobject-introspection-1.78.1-1.bottle_manifest.json
==> Installing dependencies for gobject-introspection: pkg-config
rm /opt/homebrew/bin/pkg-config
rm /opt/homebrew/share/aclocal/pkg.m4
rm /opt/homebrew/share/doc/pkg-config
rm /opt/homebrew/share/man/man1/pkg-config.1
==> Installing gobject-introspection dependency: pkg-config
==> Downloading https://ghcr.io/v2/homebrew/core/pkg-config/manifests/0.29.2_3
Already downloaded: /Users/benallen/Library/Caches/Homebrew/downloads/ac691fc7ab8ecffba32a837e7197101d271474a3a84cfddcc30c9fd6763ab3c6--pkg-config-0.29.2_3.bottle_manifest.json
==> Pouring pkg-config--0.29.2_3.arm64_sonoma.bottle.tar.gz
ln -s ../Cellar/pkg-config/0.29.2_3/bin/pkg-config pkg-config
ln -s ../../Cellar/pkg-config/0.29.2_3/share/aclocal/pkg.m4 pkg.m4
ln -s ../../Cellar/pkg-config/0.29.2_3/share/doc/pkg-config pkg-config
ln -s ../../../Cellar/pkg-config/0.29.2_3/share/man/man1/pkg-config.1 pkg-config.1
Error: No such file or directory @ rb_sysopen - /Users/benallen/Library/Caches/Homebrew/downloads/84ce7ed13999bc49b583c1b0bfd220d1bb0682ca2316d985b0f5f9e9c0a9084d--pkg-config--0.29.2_3.arm64_sonoma.bottle.tar.gz
`

heyshouzi, 2023-11-30:
> Remove the -lxt2 flag from the vvp command, if you want to generate a VCD file.

this work！！！  thanks

sachinkum0009, 2023-12-02:
> when i follow `brew install --HEAD randomplum/gtkwave/gtkwave` gtkwave can be open. but i use iverilog to compile verilog document and then use gtkwave generate wave picture. ![image](https://private-user-images.githubusercontent.com/41294649/286975927-d5f5b666-06a0-4e58-93ad-69f7093d46b9.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE1NTE0MTMsIm5iZiI6MTcwMTU1MTExMywicGF0aCI6Ii80MTI5NDY0OS8yODY5NzU5MjctZDVmNWI2NjYtMDZhMC00ZTU4LTkzYWQtNjlmNzA5M2Q0NmI5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjAyVDIxMDUxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcxZTg0YjQ1YzYxZmU4MzhiZDk5NmRiODlhODhiNTY1MzlkMmVhMTM2ZmU5ZTY3OTZiMGFjYWNiMDgxMWNmMWEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.UBCX4WgBjmZ2GE1_yXGz01HOMa8fgMtuzFEpVC9s3yI)
> 
> then occur problem as follow <img alt="截屏2023-11-30 23 11 10" width="553" src="https://private-user-images.githubusercontent.com/41294649/286975677-18e9f94c-bc99-4336-8adf-6e1ec269cda1.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDE1NTE0MTMsIm5iZiI6MTcwMTU1MTExMywicGF0aCI6Ii80MTI5NDY0OS8yODY5NzU2NzctMThlOWY5NGMtYmM5OS00MzM2LThhZGYtNmUxZWMyNjljZGExLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFJV05KWUFYNENTVkVINTNBJTJGMjAyMzEyMDIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjMxMjAyVDIxMDUxM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWRhMzQwY2Q1ZDg5ODM5NWVkNGJjNDNiZjlkNzJhZDRmNmU5NDc3M2M5ZDQ1MDhhZjcyMjdlZWEwMTMxZTQyNWUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.fo8uNcWj4lJy65uNEcK-mE_6Qp_g7iJxRYO6ajQUbeY">
> 
> how I can fix it?
> 
> I guess maybe I made a mistake in generating the format.
> 
> (this my first ask questions if I give less information,please let know).

Thanks a lot. ❤️

sachinkum0009, 2023-12-02:
> @Khizarkk7 @terriblefire There is no working release for now. you can install it using homebrew and my tap with `brew install --HEAD randomplum/gtkwave/gtkwave`

Thanks a lot :heart:

Young-Angelo, 2023-12-10:
Will a working App image be provided in anytime future?

rfuest, 2023-12-10:
> Will a working App image be provided in anytime future?

I would like to provide a non homebrew version of GTKWave for the next major release, but a new major release won't happen anytime soon.

As I have state before I don't have access to a MacOS machine and cannot work on improving support for that OS. But if someone from the community wants to step in we could create nightly builds using CI. Or perhaps even setup CI to create a new release from the GTKWave 3.3 LTS branch, which could then be published on the SourceForge page, but I have no idea about the current state of MacOS support in the LTS branch.

Young-Angelo, 2023-12-10:
> But if someone from the community wants to step in we could create nightly builds using CI.

Are those MacOS specific patches submitted by @randomplum added to main branch?

ashleycesca, 2023-12-12:
Hello everyone, thank you so much for the help and @randomplum for your detailed instructions. I can currently use gtkwave with Sonoma 14.1.12. 

However, I can't use gtkwave from my applications folder, I still get the same error that it's incompatible with macOS 14. I am only able to use it from the following path 

/opt/homebrew/Cellar/gtkwave/HEAD-d793aa1_2/bin

It works so that's good but not sure why I cannot use the it from the applications folder.

ktywb, 2023-12-18:
> Hello everyone, thank you so much for the help and @randomplum for your detailed instructions. I can currently use gtkwave with Sonoma 14.1.2.
> 
> However, I can't use gtkwave from my applications folder, I still get the same error that it's incompatible with macOS 14. I am only able to use it from the following path
> 
> /opt/homebrew/Cellar/gtkwave/HEAD-d793aa1_2/bin
> 
> It works so that's good but not sure why I cannot use the it from the applications folder.

[gtk_wave.zip](https://github.com/gtkwave/gtkwave/files/13700192/gtk_wave.zip)

Maybe you could try this.
I used the method provided by @randomplum to successfully install gtkwave. and packaged it.

ktywb, 2023-12-18:
> Hello everyone, thank you so much for the help and @randomplum for your detailed instructions. I can currently use gtkwave with Sonoma 14.1.2.
>
> However, I can't use gtkwave from my applications [...]

Maybe you could try this.
I used the method provided by @randomplum to successfully install gtkwave. and packaged it.

[gtk_waveapp.zip](https://github.com/gtkwave/gtkwave/files/13700240/gtk_waveapp.zip)

ktywb, 2023-12-18:
> Hello everyone, thank you so much for the help and @randomplum for your detailed instructions. I can currently use gtkwave with Sonoma 14.1.2.
>
> However, I can't use gtkwave from my applications [...]

1. Create Application Folder Structure:
    Create a new folder named "gtkwave".
    Inside this folder, create a subfolder named "Contents".
    Within the Contents folder, create two more subfolders: "MacOS" and "Resources".

2. Place Unix Executable File:
Place your Unix executable file inside the Contents/MacOS folder. Ensure the file has execute permissions (use the chmod +x command).

3. Create Info.plist File:
Create an XML file named Info.plist inside the Contents folder. 

4. Add “.app” Extension to the “gtkwave” Folder:

I directly used the Info.plist and the contents of the Resources folder from the package downloaded from https://gtkwave.sourceforge.net/gtkwave.zip. After doing this, it should be able to run.

I tried to upload the application I had packaged, but I found that after downloading the uploaded file, the application would not run.

neulxf, 2024-01-03:
I use the Mac ports install the gtkwave, it is ok
`sudo port install gtkwave`

Arnav3094, 2024-01-10:
> @Khizarkk7 @terriblefire There is no working release for now. you can install it using homebrew and my tap with `brew install --HEAD randomplum/gtkwave/gtkwave`

So I ran this, was a pretty long install for some reason, but then I dont see any gtkwave application, and apart from that, pfa the screenshot of a terminal session after the install.
[image]

Please let me know if I executed something incorrectly, and I apologise for any mistakes. Any help is greatly appreciated!

neulxf, 2024-01-11:
I used the `ports` to install gtkwave， its ok

---- Replied Message ----
| From | ***@***.***> |
| Date | 01/11/2024 04:33 |
| To | ***@***.***> |
| Cc | ***@***.***>***@***.***> |
| Subject | Re: [gtkwave/gtkwave] Unable to launch GTKWave on macOS Sonoma (Issue #250) |

@***@***.*** There is no working release for now. you can install it using homebrew and my tap with brew install --HEAD randomplum/gtkwave/gtkwave

So I ran this, was a pretty long install for some reason, but then I dont see any gtkwave application, and apart from that, pfa the screenshot of a terminal session after the install.
Screenshot.2024-01-11.at.2.02.32.AM.png (view on web)

Please let me know if I executed something incorrectly, and I apologise for any mistakes. Any help is greatly appreciated!

Young-Angelo, 2024-01-18:
Can a official brew be provided as of now?

sachinkum0009, 2024-01-18:
> So I ran this, was a pretty long install for some reason, but then I dont see any gtkwave application, and apart from that, pfa the screenshot of a terminal session after the install.

Once you install `gtkwave` using the `brew install --HEAD randomplum/gtkwave/gtkwave` then you can run this command, `$gtkwave myFifoTb_wave.vcd` to open the gtkwave window.

[image]

randomplum, 2024-02-04:
> Can a official brew be provided as of now?

I will try to do it when the version 3.4.0 is released. I also plan to work on making mac .app build when it's out, but this depends heavily on how much time I'll have available then.

mohitsaini50, 2024-02-10:
@randomplum, 

i am getting empty installation error while running the commands . 
brew uninstall gtkwave
brew untap randomplum/gtkwave
brew install --HEAD randomplum/gtkwave/gtkwave

[image]

heinhtetyang, 2024-03-13:
Why can't mine still work?
[image]
It has taken me a week. crying....my brain is going messy.
please Can any private message me.
Maybe I misunderstood the article.

Tond4, 2024-04-12:
> since it's head only formula tap to update you need to run: `brew uninstall gtkwave` `brew untap randomplum/gtkwave` `brew install --HEAD randomplum/gtkwave/gtkwave`
> 

why do I need a password for this repo?
[image]

rfuest, 2024-04-12:
Your command contains a typo. The screenshot shows that you entered `randomlum` instead of `randomplum`.

SKSADIRUDDIN, 2024-04-27:
should I run these command in terminal after the error shows up ? 
1.brew uninstall gtkwave
2.brew tap randomplum/gtkwave
3.brew install --HEAD randomplum/gtkwave/gtkwave

seanchannel, 2024-05-03:
I was able to run gtkwave just fine on macOS Sonoma 14.4.1 (23E224)

$ sudo port install gtk-osx-application-gtk2
$ sudo port install tk +quartz
$ sudo port install gtkwave

I think the problem is that Tk is not always configured with quartz support by default in either homebrew or macports. When you want to install a gtk app with quartz support, look into the 'port info' for each dependency as they may not build with quartz by default and you have to add the flag as above for tk.

Young-Angelo, 2024-05-04:
> I was able to run gtkwave just fine on macOS Sonoma 14.4.1 (23E224)
>
>
>
> $ sudo port install gtk-osx-application-gtk2
>
> $ sudo port install tk +quartz
>
> $ sudo port install gtkwave
>
>
>
> I think the [...]

Are you running an intel or m-based mac?

seanchannel, 2024-05-05:
This is Apple M2 @Young-Angelo and TBH I'm not sure it's completely working (non-expert here). It kind of looks like it might not be completely working with my sample code. Looks like I'll have to try the homebrew build..
`brew install --HEAD randomplum/gtkwave/gtkwave`

ian-l-kennedy, 2024-05-15:
Cannot get gtkwave on Mac M3 Max
 
==> Installing randomplum/gtkwave/gtkwave dependency: meson
==> Downloading https://ghcr.io/v2/homebrew/core/meson/manifests/1.4.0
Already downloaded: /Users/ian.kennedy/Library/Caches/Homebrew/downloads/57b6f14cf4921ae4b59488d13bfbedbc28dbeba1b71d1b83e33e4385a67b6d44--meson-1.4.0.bottle_manifest.json
==> Pouring meson--1.4.0.all.bottle.tar.gz
🍺  /opt/homebrew/Cellar/meson/1.4.0: 251 files, 3.7MB
==> Installing randomplum/gtkwave/gtkwave --HEAD
==> meson setup build -Dintrospection=false
==> meson compile -C build --verbose
==> meson install -C build
Error: undefined method `bottle_specification' for nil:NilClass
/opt/homebrew/Library/Homebrew/formula.rb:2582:in `bottle_hash'
/opt/homebrew/Library/Homebrew/sbom.rb:37:in `create'
/opt/homebrew/Library/Homebrew/formula_installer.rb:834:in `finish'
/opt/homebrew/Library/Homebrew/upgrade.rb:238:in `install_formula'
/opt/homebrew/Library/Homebrew/install.rb:350:in `install_formula'
/opt/homebrew/Library/Homebrew/install.rb:301:in `block in install_formulae'
/opt/homebrew/Library/Homebrew/install.rb:300:in `each'
/opt/homebrew/Library/Homebrew/install.rb:300:in `install_formulae'
/opt/homebrew/Library/Homebrew/cmd/install.rb:297:in `run'
/opt/homebrew/Library/Homebrew/brew.rb:92:in `<main>'
Please report this issue:
  https://docs.brew.sh/Troubleshooting

rfuest, 2024-05-15:
I don't have access to a Mac and can only give you very limited help on this problem. I think the actual build of GTKWave succeeds and only the packaging step is broken.

The undefined method in the error message is inside a file that is shipped with homebrew and not the GTKwave formula. Which version of homebrew are you running and do other packages build as expected? You could try to create a log with the `brew gist-logs` method that is mentioned in the troubleshooting link, perhaps that contains more information to narrow in on the problem.

maxekman, 2024-05-19:
Found a related upstream issue: https://github.com/Homebrew/brew/issues/17333

azharlightelligence, 2024-06-14:
For MacOS Sonoma, and Apple M1 Chip: 
Execute these commands step by step (and check for the messages after each installation incase something more required):
```
$ sudo port install gtk-osx-application-gtk2
$ sudo port install gtk2 +quartz
$ sudo port install gtkwave
```

SayatAbdikul, 2024-06-21:
> For MacOS Sonoma, and Apple M1 Chip (Running Fine!!): Execute these commands step by step (and check for the messages after each installation incase something more required):
> 
> ```
> $ sudo port install gtk-osx-application-gtk2
> $ sudo port install gtk2 +quartz
> $ sudo port install gtkwave
> ```
This error occurs after the first command:
`Error: Current platform "darwin 23" does not match expected platform "darwin 22"
Error: If you upgraded your OS, please follow the migration instructions: https://trac.macports.org/wiki/Migration
OS platform mismatch
    while executing
"mportinit ui_options global_options global_variations"
Error: /opt/local/bin/port: Failed to initialize MacPorts, OS platform mismatch`

[26 more replies]