gtkwave/gtkwave #253: Transaction filter removes original trace in viewer

guymcswain, 2023-09-28:
I'm writing a stub for a transaction filter that simply outputs the following:
```
$next
$name Decoded Data
#0
#10000 Hello Transaction Filter!
#20000
$finish
```
I've inserted a blank line under the highlighted trace to apply the filter.  The highlighted signal remains in the signal pane but the waveform is replaced with a flat line colored yellow/gold.

I would like to keep the original trace in the viewer to compare against my filtered output.  Do I need to send the original trace vcd back along with the decoded data?

PS Thanks for this wonderful tool!

gtkwave, 2023-09-28:
If you want to keep the original trace for comparison, you need to import it like normal, except don't turn on the transaction filter for that specific trace as you would for the other one.

In many cases, the original trace can provide incomprehensible data, which is why it's not preserved.  There gets to be a point where providing an option for everything gets out of hand.

-Tony

guymcswain, 2023-09-28:
Thanks.  Very flexible tool.  I solved by highlighting my original signal then edit/copy, edit/paste and finally run the transaction filter on the copied trace and voila.

closed 2023-09-28 by guymcswain