gtkwave/gtkwave #256: Update documenation

rfuest, 2023-09-29:
Some parts of the documentation aren't accurately reflecting the current development version of GTKWave any more and need to be updated. This issue is intended to track outdated parts of the documentation.

- [ ] The build instructions (**README** /  **Docs**)
- [ ] The list of supported filetypes (**README** /  **Docs** / **man** / GitHub About)
- [ ] Hierarchy Search was removed (**Docs**)
- [ ] SST can no longer be a separate window  (**Docs**)
- [ ] Some screenshots still show the old interface without a toolbar  (**Docs**)
- [ ] The LXT file format description can be removed  (**Docs**)
- [ ] Some menu options have been removed  (**Docs**)

FurryAcetylCoA, 2024-04-24:
I noticed a discrepancy between the documentation and the actual behavior of gtkwave in the section titled `Transaction Filters example` on pages 64-65.
The example provided in the documentation is as follows:
```
$comment data_start 0x124c0798 $end
$comment val[7:0] $end
$timescale 1ms $end
$comment min_time 0 $end
$comment max_time 348927 $end
$comment max_seqn 1 $end
$scope module top $end
$comment seqn 1 top.val[7:0] $end
$var wire 8 1 val[7:0] $end
$upscope $end
$enddefinitions $end
$dumpvars
#0
b10000000 1
#1
b10000101 1
#2
b10001010 1
…
```
However, the actual output from gtkwave is different ( differences are indicated by  `**` , comments are indicated by `//`  (which do not exists in the actual outputs)):
```
$comment data_start 0x124c0798 $end
$comment **name** val[7:0] $end // vcd_saver:485
$timescale 1ms $end
$comment min_time 0 $end
$comment max_time 348927 $end
$comment max_seqn 1 $end
$comment args "0" $end // vcd_saver:517 only exists if args are given
$scope module top $end
$comment seqn 1 top.val[7:0] $end
$var wire 8 1 val[7:0] $end
$upscope $end
$enddefinitions $end
**#0** // vcd_saver.c: 622
**$dumpvars** // vcd_saver.c: 627
b10000000 1
**$end** // vcd_saver.c: 619 
#1
b10001001 1
#2
b10001010 1
…
```
Also, I checked the 18.2.3.12 and 18.2.4 in IEEE.1364-2005. The output from gtkwave is correct.

Given that the documentation format is in ODT, a binary format.
I am unable to submit a pull request to correct this.

rfuest, 2024-04-24:
> Given that the documentation format is in ODT, a binary format.
> I am unable to submit a pull request to correct this.

There is an open issue to migrate the docs to markdown #144, but nobody is working on finishing that at the moment.
You should still be able to open the ODT file with LibreOffice and create a PR.