gtkwave/gtkwave #268: Ubuntu 22.04.3 LTS Build failure

keesj, 2023-10-13:
Hello,

I am trying to build gtkwave using the new meson based build system on ubuntu 22.04. I installed most of the dependencies by adding
deb-src sources in my source.list  e.g. 
```
$ cat /etc/apt/sources.list
deb http://archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
deb http://archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
deb http://security.ubuntu.com/ubuntu jammy-security main restricted universe multiverse
deb-src http://archive.ubuntu.com/ubuntu jammy main restricted universe multiverse
deb-src http://archive.ubuntu.com/ubuntu jammy-updates main restricted universe multiverse
deb-src http://security.ubuntu.com/ubuntu jammy-security main restricted universe multiverse
```

After a few more build failure  and install of package (like libgirepository1.0-dev ) I can run meson compile in the build directory.

I currently is failing in compiling **gw-color.c** that was modified about 2 weeks ago.

```
[1/25] Compiling C object lib/libgtkwave/src/libgtkwave.so.p/gw-color.c.o
FAILED: lib/libgtkwave/src/libgtkwave.so.p/gw-color.c.o
ccache cc -Ilib/libgtkwave/src/libgtkwave.so.p -Ilib/libgtkwave/src -I../lib/libgtkwave/src -I. -I.. -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c99 -O2 -g -D_GNU_SOURCE -DGLIB_VERSION_MIN_REQUIRED=GLIB_VERSION_2_64 -DGLIB_VERSION_MAX_ALLOWED=GLIB_VERSION_2_64 -fPIC -MD -MQ lib/libgtkwave/src/libgtkwave.so.p/gw-color.c.o -MF lib/libgtkwave/src/libgtkwave.so.p/gw-color.c.o.d -o lib/libgtkwave/src/libgtkwave.so.p/gw-color.c.o -c ../lib/libgtkwave/src/gw-color.c
../lib/libgtkwave/src/gw-color.c:11:5: error: initializer element is not constant
   11 |     { \
      |     ^
../lib/libgtkwave/src/gw-color.c:19:5: note: in expansion of macro ‘X11_CONST’
   19 |     X11_CONST("alice blue", 240, 248, 255),
      |     ^~~~~~~~~
../lib/libgtkwave/src/gw-color.c:11:5: note: (near initialization for ‘X11_COLORS’)
   11 |     { \
      |     ^
../lib/libgtkwave/src/gw-color.c:19:5: note: in expansion of macro ‘X11_CONST’
   19 |     X11_CONST("alice blue", 240, 248, 255),
      |     ^~~~~~~~~
../lib/libgtkwave/src/gw-color.c:20:5: warning: missing initializer for field ‘color’ of ‘X11Color’ [-Wmissing-field-initializers]
   20 |     X11_CONST("AliceBlue", 240, 248, 255),
      |     ^~~~~~~~~
../lib/libgtkwave/src/gw-color.c:6:13: note: ‘color’ declared here
    6 |     GwColor color;
      |             ^~~~~
../lib/libgtkwave/src/gw-color.c:21:5: warning: missing initializer for field ‘color’ of ‘X11Color’ [-Wmissing-field-initializers]
   21 |     X11_CONST("antique white", 250, 235, 215),
      |     ^~~~~~~~~
../lib/libgtkwave/src/gw-color.c:6:13: note: ‘color’ declared here
    6 |     GwColor color;
      |             ^~~~~
../lib/libgtkwave/src/gw-color.c:22:5: warning: missing initializer for field ‘color’ of ‘X11Color’ [-Wmissing-field-initializers]
   22 |     X11_CONST("AntiqueWhite", 250, 235, 215),
      |     ^~~~~~~~~
```

```
:~/projects/gtkwave/build$ gcc --version
gcc (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Greetings

keesj, 2023-10-13:
It appears the compiler does not like the cast

```

diff --git a/lib/libgtkwave/src/gw-color.c b/lib/libgtkwave/src/gw-color.c
index b37061c..2da88b2 100644
--- a/lib/libgtkwave/src/gw-color.c
+++ b/lib/libgtkwave/src/gw-color.c
@@ -9,7 +9,7 @@ typedef struct
 #define X11_CONST(name, r, g, b) \
     (X11Color) \
     { \
-        name, (GwColor) \
+        name,  \
         { \
             r / 255.0, g / 255.0, b / 255.0, 1.0 \
         } \
@@ -912,4 +912,4 @@ const GwColor *gw_rainbow_color_to_color(GwRainbowColor self)
     };

     return &RAINBOW_COLORS[self];
-}
\ No newline at end of file
+}
```

rfuest, 2023-10-13:
I had also recently encountered this issue on GCC 12. It should be fixed now: #270.

closed 2023-10-13 by rfuest