gtkwave/gtkwave #269: Ubuntu WSL2 Missing dragging handle bars

keesj, 2023-10-13:
I noticed that when running gtkwave under WSL2 most of the time I am missing the handle bars that allow to resize the panes.
It is still possible to move them but it is a bit difficult (also the mouse cursor does not change in such situation)

[image]

Sometime I have see the bars  and I also tried different themes but this does not change the behaviour.

keesj, 2023-10-13:
They do show up in 3.3.104

[image]

rfuest, 2023-10-13:
The 3.3.104 in your screenshot seems to use GTK 2, which used a very wide handle. I've already made the draggable parts easier to identify in my initial work on a GTK 4 port, but I'll see if I can backport these changes to the GTK 3 version of GTKWave.

> ... (also the mouse cursor does not change in such situation)

That's not good, I haven't encountered this issue before. Does the cursor change correctly in other GTK 3 application in WSL? You could for example try `gtk3-widget-factory` from the `gtk-3-examples` package. "Page 2" in the demo application contains some of the same widgets we use in GTKWave in the bottom center.

guymcswain, 2023-10-13:
I have gtkwave running on WSL2 installed from a fresh Ubuntu-22.04 distro.  Version 3.3.104 was installed and don't have any issue with the handle bars as mentioned here.  But the comment about the screenshot resembling an older GTK library led me to check:

```bash
Ubuntu-22.04:.../txook/gtkwave$ dpkg -l libgtk* | grep -e '^i' | grep -e 'libgtk-*[0-9]'
ii  libgtk2.0-0:amd64 2.24.33-2ubuntu2 amd64        GTK graphical user interface library - old version
ii  libgtk2.0-bin     2.24.33-2ubuntu2 amd64        programs for the GTK graphical user interface library
ii  libgtk2.0-common  2.24.33-2ubuntu2 all          common files for the GTK graphical user interface library
```
So, I'd like to properly install gtkwave with (at least) GTK 3.  But given #256 - build instructions out of date - how should I do this?

rfuest, 2023-10-14:
> So, I'd like to properly install gtkwave with (at least) GTK 3. But given #256 - build instructions out of date - how should I do this?

They are out of date for the latest GIT version. What you should do depends on what you want to achieve. If you only want to use a recent GTKWave version and don't want to deal with any bugs that might have been introduced in the development branch I would suggest you stick with GTKWave LTS 3.3. You can either run the Flatpak version (https://flathub.org/apps/io.github.gtkwave.GTKWave) or compile it yourself from a relase on SF (https://sourceforge.net/projects/gtkwave/files/). The build instructions in the README are still valid for the SF version.

If you want to try the development version you'll need to use `meson` to compile GTKWave:
```sh
sudo apt install build-essential meson gperf flex desktop-file-utils libgtk-3-dev libbz2-dev libjudy-dev
git clone https://github.com/gtkwave/gtkwave.git
cd gtkwave
meson setup build
cd build
meson compile
sudo meson install
```

guymcswain, 2023-10-14:
Thank you for the detailed response.  I'll go down the SF path.  To confirm, as of today's date, the LTS version would be 3.3.114 (the big green button), even though the latest files are 3.3.117?

keesj, 2023-10-14:
> That's not good, I haven't encountered this issue before. Does the cursor change correctly in other GTK 3 application in WSL? You could for example try `gtk3-widget-factory` from the [...]

Hello, the gtk3-widget-factory show the same behaviour that there is no mouse hint when I hover over the "bars". Hence this is probably not an issue with gtkwave but something elsewhere. 

[image]

rfuest, 2023-10-14:
Maybe the icon theme, which contains the cursor icon isn't installed. I'm not sure which package contains the correct files, but you could try to install `adwaita-icon-theme-full`.