gtkwave/gtkwave #278: Reaload button not fully working

keesj, 2023-10-17:
The reload functionality on  f11428052b3ca572957b4face9ba8c937d9408b4 (current head) is broken

Steps to reproduce
Given the following verilog code
```
`timescale 1ns/1ps
`default_nettype none

module clk_tb;
  reg clk =1'b0;

  always begin
    #5 clk = ~clk;
  end

  initial begin
    $dumpfile("clk_tb.vcd");
    $dumpvars(0,clk_tb);
    #50;
    $finish;
  end

endmodule
```

Generate vcd code  using iverilog
Open a file and add the clock line
[image]

Press the reload button

[image]

Is appears that the signal is still visible (in the top left corner) but zooming has little effect

rfuest, 2023-10-17:
I did take a quick look at this issue and it is caused by a global variable that is overwritten by the reload. The affected variable will be removed as part of the waveform drawing refactoring, which will make the code more robust against issues like this in the future. I'll try to extract the part that fixes this issue from the refactoring project and merge it soon.
As a temporary workaround you can resize the GTKWave window after the reload to restore the correct waveform display.

ams, 2023-11-22:
Would be really nice to back port this fix, since it makes gtkwave very hard to use in its current state.  Anything one can do to help?

rfuest, 2023-11-22:
I had hoped to backport a better fix for this problem from my redrawing rewrite, but decided that this would be too much effort. But I've just pushed a temporary hack that should fix reloading in the meantime.