gtkwave/gtkwave #280: Enormous RAM usage with big wave files.

albydnc, 2023-11-01:
I have a wavefile in GHW format that is ~300 MB in size.
I have tried to open it with gtkwave and the tool hangs requesting tons of memory ( I have a 16 GB machine and it takes 20 GB requiring the os to swap).
It never opens the GUI.

rfuest, 2023-11-01:
All traces in a GHW file are loaded at once and converted into a format that is optimized for displaying in GTKWave, which causes the large memory consumption. If you can use FST files instead of GHW files you should try to use that format instead. Traces in FST files are lazily loaded when they are needed.

albydnc, 2023-11-01:
Unfortunately, the simulator I'm using is not able to export some signals into the FST format (see [issue on GHDL](https://github.com/ghdl/ghdl/issues/2527#issue-1973128643) ), so I need to use GHW.

rfuest, 2023-11-02:
I'm afraid there might not be much we can do to fix this in the short term. In the long term it will be possible to improve GTKWave's memory consumption for GHW files, but these changes will need to wait until the current refactoring is finished.

Can you provide the GHW file, or one with similar characteristics, for further testing? My test file did "only" consume about ~15 times the GHW filesize as the peak memory consumption. Your file is much worse than that and it would be interesting to check if the high memory consumption is just caused by the expected inefficiencies or if something else is wrong.

albydnc, 2023-11-02:
Thanks for the help! Here is the GHW file in question. https://cernbox.cern.ch/s/GBA8otWWO0MgeqR

rfuest, 2023-11-03:
Thanks for the file, getting access to actual dump files helps a lot to improve GTKWave.

The GHW file contains 35.649.163 signals, which causes GTKWave to use ~14 GB of memory just to store the hierarchy. The GHW loader seems to create one hierarchy entry for each bit in the design, which consumes a lot of memory. A huge chunk of this memory is consumed by storing the full name of each of the 35 million signals. Fixing this will require some major changes to the GHW loader and/or the way GTKWave stores the hierarchy.

But there are more issues than that. After the hierarchy is built the memory consumption doesn't increase anymore and no waveform data seems to be imported. I haven't investigated further yet, but this might also be an issue with the GHW library. The `ghwdump` tool, which is bundled with GHDL, did also seem to hang when I try to get more information about the GHW file. But perhaps it is just slow and I didn't wait long enough for the program to terminate. What is the end time you used for the simulation? I did let `ghwdump -T` run for quite some time and the time was at ~30 ms when I terminated the process.

rfuest, 2023-11-05:
I did some more tests and the GHW import library has a serious performance issue. There is a linear search over the signals in `ghw_read_cycle_cont`: https://github.com/gtkwave/gtkwave/blob/d793aa1bef876d56233e80f5b8fc3fd704ef8c00/lib/libghw/libghw.c#L1525-L1530
Without this code the runtime of `ghwdump -T` did reduce from many minutes (I didn't wait long enough to get an actual measurement) to 2-3 seconds. This will need to be fixed before GTKWave can be optimized to handle larger GHW files.

albydnc, 2023-11-06:
Thank you for the analysis. I will try to help to improve it.

soenke-brandt, 2023-11-30:
> I did some more tests and the GHW import library has a serious performance issue. There is a linear search over the signals in `ghw_read_cycle_cont`:

See also: 
[Issue](https://gitlab.fel.cvut.cz/canbus/ctucanfd_ip_core/-/issues/284), [Patch](https://gitlab.com/canfd/server-tools/-/commit/7d0e8b6d01b3406b8167f7fcbf44f4e714e7c7cf)

stefanlippuner, 2024-01-19:
Hi all!

Here's an update regarding libghw: A patch to libghw has been merged upstream in https://github.com/ghdl/ghdl/pull/2585. It removes the linear search in all or most cases, which reduces the load time of this log to a few seconds in ghwdump. Memory consumption of gtkwave grows to about 29 GB, but I'm able to open the file now (takes 1-2 min to load). There's also some weird behaviour, e.g., signals change name after inserting them to the wave window. I'm not sure what exactly goes wrong there, but it doesn't seem to be related to the `ghw_read_cycle_cont` changes.

rfuest, 2024-01-20:
Thanks for the update. I have some ideas how memory consumption could be reduced in the future, but the first step is to finish my refactoring of the loaders. At the moment I'm trying to limit the changes to the loaders to keep the risk of accidentally breaking anything smaller. But I've already added some automated tests as part of this first refactoring pass, which should help with bigger changes in the future.

> Memory consumption of gtkwave grows to about 29 GB, but I'm able to open the file now (takes 1-2 min to load).

Which GTKWave version did you use and on which OS? I had no luck opening the file in either the master branch or the LTS version (both with the libghw patch applied). Loading gets stuck in `ghw_sortfacs` in both cases.

> There's also some weird behaviour, e.g., signals change name after inserting them to the wave window. I'm not sure what exactly goes wrong there, but it doesn't seem to be related to the `ghw_read_cycle_cont` changes.

I have encountered some issues with the GHW loader, but not this exact problem yet and my automated test still work after applying the libghw patch. Did you test your build of GTKWave with other GHW files?

stefanlippuner, 2024-01-20:
> At the moment I'm trying to limit the changes to the loaders to keep the risk of accidentally breaking anything smaller.

That sounds quite reasonable. I may still do some poking around (I have a encountered a different stack overflow issue when loading my own ghw file), I hope that's ok. Do you plan to stick with C, or e.g. switch to C++? I feel that some of the standard C++ features (just vectors in my case) would be quite useful.

> Which GTKWave version did you use and on which OS? I had no luck opening the file in either the master branch or the LTS version (both with the libghw patch applied). Loading gets stuck in ghw_sortfacs in both cases.

I tested the changes on the current master, e.g. using ebd66120018b78292f5cba56345d016a7033f2e5, on Linux Mint 21.3. I used the default (debugoptimized) build `rm -rf build && meson setup build && meson compile -C build`. Loading takes quite a while, about 90s on my Ryzen 7700X, and a good chunk of it is in `ghw_sortfacs`. The sigcmp_2 changes from the patch mentioned above (https://github.com/gtkwave/gtkwave/commit/dcb7041bef682c28fb8219fac20eaaa22a877ccd) seem to help with that (60s instead of 90s) but I don't fully understand those changes.

> I have encountered some issues with the GHW loader, but not this exact problem yet and my automated test still work after applying the libghw patch. Did you test your build of GTKWave with other GHW files?

I did test it on some other files and those worked fine.

rfuest, 2024-01-20:
> I hope that's ok. Do you plan to stick with C, or e.g. switch to C++? I feel that some of the standard C++ features (just vectors in my case) would be quite useful.

There is no plan to switch to C++. This is only my opinion, but I've never liked C++ and would rather switch to Rust. But at the moment sticking with C should be the way to go.

The refactored code does rely much more on the types provided by the GLib library, which should make the C code easier to read. GLib contains GArray/GPtrArray as possible replacements for a C++ vector.

> I tested the changes on the current master, e.g. using [ebd6612](https://github.com/gtkwave/gtkwave/commit/ebd66120018b78292f5cba56345d016a7033f2e5), on Linux Mint 21.3. I used the default (debugoptimized) build `rm -rf build && meson setup build && meson compile -C build`. Loading takes quite a while, about 90s on my Ryzen 7700X, and a good chunk of it is in `ghw_sortfacs`. The sigcmp_2 changes from the patch mentioned above ([dcb7041](https://github.com/gtkwave/gtkwave/commit/dcb7041bef682c28fb8219fac20eaaa22a877ccd)) seem to help with that (60s instead of 90s) but I don't fully understand those changes.

Very strange, I have a very similar system configuration (Ryzen 7700X with Arch Linux), but couldn't get it to work. I'll need to retest this later.

> I did test it on some other files and those worked fine.

OK, thanks for checking. I'll look into the issue when I get the big GHW file opened.

stefanlippuner, 2024-01-20:
> There is no plan to switch to C++. This is only my opinion, but I've never liked C++ and would rather switch to Rust. But at the moment sticking with C should be the way to go.
>
> The refactored code [...]

Rust would would of course also work to get vectors but that would be a _slightly_ bigger effort. Thanks for pointing me to glib! It should probably also be sufficient for the stack overflow fix. I'll see if I can implement it with that.

I also tried opening the ghw file with my changes rebased on the LTS branch. It also took much longer (I terminated it in `ghw_sortfacs`, after `wave_heapsort`) to open and consumed even more RAM.

rfuest, 2024-01-20:
If you want to work on the GHW loader, which would be very much appreciated, I would suggest you base your work on my ongoing effort to refactor the loaders. I've opened a draft PR #302 for my work in progress branch. The GHW loader is now located in `lib/libgtkwave/src/gw-ghw-loader.c`.

tbybell, 2024-01-20:
The sigcmp_2() function parses out numbers and does a numeric sort vs a text one for the number encountered.  In this way if you have a[12] > a[2] rather than the ASCII "1" being less than ASCII "2".  There are lots of ways to optimize this.  GHW will see this a lot more than Verilog because of the split up bits coming from GHW's reader library.

At my day job I solved this "stuck on ghw_sortfacs()"-type problem involving netlist symbols by converting numbers to a fixed length binary representation that *would* sort fine with strcmp() like algorithms then after sort was done, I converted them back.  You can have your cake and eat it!  Note that for how GHW stores signals bit blasted, I know for certain that instead of sorting, that building into a (radix) trie IS considerably faster than sifting through redundant, lengthy prefixes on flattened hierarchical names over and over.  You are getting burned so hard on signal comparisons due to your TLB accesses missing it's not even funny.  If you think you have a lot of signals to sort, try sorting the scandump flop names in an 8-way modern processor design.  This can be done in considerably less than 60 seconds.  :-)

Optimal method is breaking down the hierarchies into trees and sorting subtrees such as the FST uses.  GHW already provides this info.  I use Verilog on my day job so I never optimized the VHDL code in gtkwave...  There are a lot of ways to solve the problem.

C++ has its place if used judiciously.  In the LTS version, I use it in a set of wrapper functions that bring C++ code into the C world because the FsdbReader library has C++ style symbol mangling.  I personally don't like C++ because it's origins are syntactic sugar for C.  Old timers will remember AT&T's cfront program that existed before C++ compilers were available.  My main concern is abuse of templates creating incomprehensible source code.

I've never used Rust, but if it's good enough for the Linux kernel, I suppose I need to get with the program.