gtkwave/gtkwave #284: When using black on white, lines are very thin ...

ams, 2023-11-19:
When using black/white mode, the lines tend to get very very thin .. making it hard to view. Is it possible to make the lines thicker?

rfuest, 2023-11-20:
You can enable the `use_fat_lines` setting in gtkwaverc to use thicker lines, but I'm not sure if that improves the usability of the black and white display much, because the grid lines also get thicker and the traces touch the grid lines.
Thin lines:
[image]

Thick lines:
[image]

But the new waveform drawing code, which I'm currently working on, will be more flexible and will also improve the black and white mode. Here is a WIP screenshot of the new code:
[image]
Unfortunately this isn't available in the GIT repo yet and it will probably take some time to get the rewrite finished.

ams, 2023-11-20:
The "fat" line one looks quite messy.  :-(. But the last one I really do like, looking forward to that!  If you are looking for testers, I'm all in.

Thank you for working on GTKwave, it is a wonderful little program!