gtkwave/gtkwave #285: Freeze when viewing large VCD files

Danacus, 2023-11-20:
Whenever I open a large VCD file (~50MB) with a lot of signals, GTKWave freezes when I try to zoom or scroll through the waves.

Steps to reproduce:
- Open [test.vcd.gz](https://github.com/gtkwave/gtkwave/files/13413306/test.vcd.gz) in GTKWave
- Insert the first 100 signals in the waves view
- Try to zoom out as far as possible until GTKWave freezes

Workaround: convert the VCD file to FST with `vcd2fst` and open the FST file in GTKWave. This seems to solve all the freezes.

rfuest, 2023-11-20:
Your VCD seems to be the cause of this issue. GTKWave does output a warning on the console, when the file is opened: `VCDLOAD | Time backtracking detected in VCD file!`. The time in a VCD file should increase monotonically, but in your file it jumps back from 4999 to 0.

But even a broken VCD shouldn't cause GTKWave to freeze. @tbybell: Why is time backtracking in VCD files only a warning and not an error?

Danacus, 2023-11-21:
Thank you for pointing this out! This is due to a bug in the code that generates this VCD file, which I had not noticed yet.