gtkwave/gtkwave #296: What is the correct way to install to a user defined prefix?

toddstrader, 2023-12-19:
I've tried variations on this theme to no avail:
```
meson setup --prefix /tmp/foo build
meson compile -C build
meson install -C build
```

The `setup` step reports:
```
  User defined options
    prefix: /tmp/foo
```

And things are installed in `/tmp/foo`.  But I get this:
```
$ /tmp/foo/bin/gtkwave
/tmp/foo/bin/gtkwave: error while loading shared libraries: libgtkwave.so: cannot open shared object file: No such file or directory
```

because:
```
$ ldd /tmp/foo/bin/gtkwave | grep libgtkwave
        libgtkwave.so => not found
```

The library does even exist in the install, but the binary doesn't know where to find it.
```
$ ls /tmp/foo/lib/x86_64-linux-gnu
girepository-1.0  libgtkwave.so  pkgconfig
```

Also, I can successfully run `build/src/gtkwave` out of the repo.

rfuest, 2023-12-21:
It would be possible to get GTKWave working in any non standard prefix by setting the rpath in the GTKWave executable, but this would only be a partial solution and isn't recommended by distributions (https://wiki.debian.org/RpathIssue).

Would manually setting `LD_LIBRARY_PATH` or adding your custom lib directory to the `/etc/ld.so.conf.d` be a satisfying solution for you? Otherwise we would need to add an option to configure whether the rpath in the executable gets set, which will allow GTKWave to run, but could cause problems in the future (e.g. with plugins) because the paths to the pkgconfig and  GObject introspection files would still not be set correctly.

toddstrader, 2023-12-21:
I'm no distro or packaging expert, but I think the argument against rpaths in that doc isn't super applicable here.  It seems to primarily be concerned with a binary and a library it depends on both being dependent on different versions of a second library.  But I'd think it should be fine to assume `gtkwave` and `libgtkwave.so` are atomic?  Again, maybe I'm not understanding the problem.

I am however not keen on `LD_LIBRARY_PATH` (or `LD_PRELOAD` for that matter) or `/etc` changes.  The latter is going to be very unpopular with our admins and hard to roll out.  And the former is dangerous as it would make it difficult to test different builds of `gtkwave`.

Again, I can build this and run it out of the repo just fine.  What I'd like to do is to install it on NFS for everyone on my team to use in a way that I can just give them the path to `gtkwave` and that's all they need to know.

I think being able to manually set the rpath (or having it be deduced from the install prefix) would work for me.  But I also don't understand the issues you raise about:
> pkgconfig and GObject introspection files

probably because I'm not using more advanced features (yet).  Is it possible to hard-code paths to those resources inside of `gtkwave` as well?

rfuest, 2023-12-21:
At the moment GTKWave and libgtkwave will be packaged as one, but this might change in the future. libgtkwave is designed to not depend on GTK and should be easier to use in headless applications. But it will take time until libgtkwave is useful on it's own.

I don't really like the rpath solution, because hardcoding paths inside executables, if other solutions exist, seems wrong. But I've decided to add an option to set the rpath anyway (#297), because I have to agree that it makes it easer to use custom GTKWave builds.