gtkwave/gtkwave #298: libfst have undefined behaviors catched by `-fsanitize=undefined`

FanShupei, 2024-01-13:
I catch several undefined behaviors with `-fsantinize=underfined` using verilator fst tracing. Since gtkwave is the upstream, I just report it here. 

The first is related to `FST_DO_MISALIGNED_OPS`
The second is an overflow bug in `lz4.c`

1. For the first bug, **I propose to completely remove `FST_DO_MISALIGNED_OPS`**, for two reasons

- misaligned pointer dereference is undefined behavior. see https://clang.llvm.org/docs/UndefinedBehaviorSanitizer.html
- misaligned pointer dereference provides no benefit than `memcpy`. In modern gcc/clang 'memcpy' is implemented as intrinsics. They generate nearly identical code for misaligned read/write, even in `-O0`. see https://godbolt.org/z/q5fr3Ksdr

2. For the second bug, the fix is actually small. However I notice that `lz4.c` is pretty outdated. Any plan to bump its version from upstream?

```diff
diff --git a/lib/libfst/lz4.c b/lib/libfst/lz4.c
index 55dc1a8..893d6dc 100644
--- a/lib/libfst/lz4.c
+++ b/lib/libfst/lz4.c
@@ -536,7 +536,7 @@ FORCE_INLINE int LZ4_compress_generic(
         }
 
         /* Catch up */
-        while ((ip>anchor) && (match+refDelta > lowLimit) && (unlikely(ip[-1]==match[refDelta-1]))) { ip--; match--; }
+        while ((ip>anchor) && (match+refDelta > lowLimit) && (unlikely(ip[-1]==(match+refDelta)[-1]))) { ip--; match--; }
 
         {
             /* Encode Literal length */
```

I'm willing to authoring a pull request. Any comments?

FanShupei, 2024-01-13:
> A PR would be appreciated. There also have been some security fixes made to the FST library in the SourceForge version recently, which should probably also be included: https://sourceforge.net/p/gtkwave/code/1651/tree//gtkwave3-gtk3/src/helpers/fst/fstapi.c?diff=50e217695fcbc91266c36e01:1650 I'm not sure why tbybell hasn't also committed these changes to the GitHub repo, but they were released as part of GTKWave 3.3.118.

So it's suggested we should first sync changes from SF to Github. But I'm pretty confused what's the relationship between SF and Github for gtkwave? Are they synced or they are two difffrent deviated branch? `libfst.c` are constant updated in SF, but its Github version hasn't be touched for several months.

> From my point of view it would be better to update the LZ4 version than to try to fix any bugs in the local version of lz4.c. But I don't know the libfst code very well and also don't know if newer versions of the LZ4 library are a simple drop in replacement for the currently used version.

That makes sense. I'm also not familiar with LZ4. I will first spend some time to figure out LZ4's backward compatibility story.

rfuest, 2024-01-13:
> So it's suggested we should first sync changes from SF to Github. But I'm pretty confused what's the relationship between SF and Github for gtkwave? Are they synced or they are two difffrent deviated branch? ...

Tony maintains the GTKWave LTS 3.3 branch on SF at the moment, but it was planned to move all GTKWave development to GitHub. I had created a `lts` branch in the GitHub repo for this purpose, but it hasn't been updated since 3.3.116. The `master` branch of the GitHub repo will become GTKWave 4 in the future.

> ... `libfst.c` are constant updated in SF, but its Github version hasn't be touched for several months.

I'm not aware of any changes to libfst on SF which aren't included in this repo, except from the changes in 3.3.118 I mentioned in my previous comment.

tbybell, 2024-01-13:
1) I just checked in the FST security fixes into git.  I forgot to push to origin master which is the only reason it wasn't done yet on GitHub.  As far as the other files that have the string TALOS in them in SF, don't worry about them for now.  From what I gather at Cisco, they have an automated formal tool the generates files used for input and they construct such malicious files in an attempt to hijack the machine through a buffer overflow, generally though malloc() being abused.  Given gtkwave's restricted usage (engineers on their own sim data), I'm not too concerned about this, however where we have a problem is the user who wants everything because he can have everything, installs the Desktop Everything version, and because gtkwave's files have mime types attached to them, there's a vector into his machine if he's careless enough to launch gtkwave.  Ralf, at some point please PM me your email so I can send you all the reports and malicious input files attached to them.  Most of the bugs are of the same flavor, so it appears they were very specific in what they were looking for.

Misaligned loads and stores being used in gtkwave are deliberate, and in fact you'll notice that the code there uses misaligned data inside structs to save on space by removing natural alignment padding between structure members:

#ifdef WAVE_USE_STRUCT_PACKING
#pragma pack(push)
#pragma pack(1)
#endif

...

#ifdef WAVE_USE_STRUCT_PACKING
#pragma pack(pop)
#endif

This is to save on memory, and yes, it's required in this day and age even on 64-bit machines, at least for those of us who have to design and debug extremely large designs such as an 8 way x86 design...  Maybe once when I get a used Threadripper, I won't need to do this.  :-)

With respect to FST, there are deliberately packed memory regions in order to be able to stuff more into the FST reformatting buffer used by the FlushContext routines, and that's why those misaligned loads exist.  If modern compilers will generate a 32-bit load, when faced with a memcpy of constant length 4, I'm fine with this because the compiler vs the target ISA is used to determine what code to generate rather than an ifdef.  I used to program on SPARC as an undergrad, and they would throw a Bus Error if a load/store was not at its naturally aligned boundary.  That's the only reason this define exists.  All modern ISAs should support this properly.

2) The lz4.c is very old.  The new one should be drop in compatible if its developer is doing things right.  I can't remember the last time I updated it.  The date on it is shown as 2015?  The only API difference I see right now is that LZ4_compress() is migrating to an LZ4_compress_default() function that expects the destination buffer length to be specified.  AFAICT, FST won't suffer any problems here until the code is updated to pass that extra argument to the new function, mainly because it already allocates a dest buffer larger in size than the theoretical max, so there's no way to defeat this unless the malloc size overflows, which was a source of most of the TALOS CVEs.

FanShupei, 2024-01-14:
Hi, tbybell. I'd like to explain more why I propose turn off `FST_DO_MISALIGNED_OPS`. I do understand FST format is designed to perform unaligned access deliberately. I'd like to say maybe **using `memcpy` is a better way to do unaligned access**.

1. 'memcpy' is more standard conforming and safe. Using misaligned access directly lies to compiler and the compiler may perform optimization assuming the address is actually aligned. It seems no compiler does it currently, but it's risk for mis-compilation in the future. And the behavior is explicitly caught by undefined sanitizer.
3. 'memcpy' has no performance cost. Using memcpy for misaligned access is a wide-used trick and the pattern is generally recognized by the compiler. If the ISA supports unaligned access, the compiler will just emit an unaligned access.
4. other projects, at least LZ4, implements several ways to do unaligned access, but using 'memcpy' is the default. see https://github.com/lz4/lz4/blob/76b07a1878db23fe19cd77d379862ea28fc8490c/lib/lz4.c#L64C1-L85C7

I'm not too much against `FST_DO_MISALIGNED_OPS`,  but I do think it shouldn't be the default. Since LZ4 is an important dependency of libfst, maybe we could borrow the strategy from LZ4, making `memcpy` way the default?

tbybell, 2024-01-14:
"If modern compilers will generate a 32-bit load, when faced with a memcpy of constant length 4, I'm fine with this because the compiler vs the target ISA is used to determine what code to generate rather than an ifdef."

I believe that with that statement that I said that I was in agreement with making the memcpy change.

tbybell, 2024-01-14:
remote: To https://github.com/gtkwave/gtkwave
   0d20555..a2979e0  master -> master

If further changes need to be made, please submit your pull request.  Thanks!

rfuest, 2024-01-15:
> 1. Ralf, at some point please PM me your email so I can send you all the reports and malicious input files attached to them.

You can find my email address in the GIT log.